RenderHeads/UnityPlugin-AVProDeckLink #3: Timecode output

AndrewRH, 2019-06-03:
It seems that I'm not getting the timecode on my output. In the "Deck Link Input" I can see the timecode getting in. But on the output there is not time code. Even if I'm connecting the output to a Hyperdeck, the timecode is always zero.
I did see that there is a setting called "Enable Time Code Capture" and I have enabled it on both the Input & the Output but still no luck. Is there additional steps I need to take to keep my timecode?

AndrewRH, 2019-06-03:
Yes that's right, we're currently not passing through the timecode on output.  Currently we only capture the timecode on input frames.  We will look into adding support for output frames too.

tetsutaroh, 2021-01-15:
I'm also interested in putting timecode in the output.
When will it be supported?

AndrewRH, 2021-01-25:
We're looking into this now and will hopefully have more information by the end of the week

RenderHeadsMrT, 2021-02-02:
Timecode output support is in development but to make the most of it we're looking at potential use cases with current options being:

1. Relative timecode from when output starts, matched to output rate
2. Pass Unity timecode through, most likely relative to app start time (rate options TBD)
3. If we have an input feed then sync Unity to it and pass its timecode through (based on current 'Synced To Input' option)

I believe 3 is the path @tetsutaroh requested and is being looked into but if anyone has any further use cases please do suggest them.

tetsutaroh, 2021-02-03:
Thank you for your response.
As you might have guessed, my hope is No.3.
I think that this function is very convenient and expands the range of applications for broadcasting.
When will it be completed? I hope it will be completed soon.

RenderHeadsMrT, 2021-02-03:
@tetsutaroh We're hoping to have a version with support for this out in a week or so (it's connected to some other work).

BlueprintBen, 2021-04-08:
Our compositing pipeline produces its own timecodes so we'd probably take #1 or more simply the ability to assign the next pushed frame's timecode. We aren't using Unity's sim clock or frame ticks.

RenderHeadsMrT, 2021-05-28:
Apologies for the delay on this - other features required attention. Options 1 and 3 from the above list have been implemented and included with the latest release.

closed 2021-05-28 by RenderHeadsMrT