RenderHeads/UnityPlugin-AVProDeckLink #4: Add HDRP/LWRP support

AndrewRH, 2019-06-03:
Currently the ApplyToMaterial/ApplyToMesh scripts don't work correctly.  This is due to materina.mainTexture not working with HDRP, so it needs to be changed eg:

					//_material.mainTexture = texture;
					_material.SetTexture("_BaseColorMap", texture);

Have to see if anything else needs changing too.

AndrewRH, 2019-09-23:
Another part about supporting HDRP that is difficult is alpha for External Keying, since HDRP doesn't support rendering to alpha channel....

krisrok, 2021-04-29:
just to clarify: is (or was) this just a tooling problem? i guess we can still manually apply the texture for it to work with the SRP-implementations? it's just texture data after all..