RenderHeads/UnityPlugin-AVProDeckLink #5: Demo 07 can not show both input and ouput

wsdfz, 2019-09-09:
I found in avpro decklink 1.7.4. Demo 07 can not show both input and ouput. It can only show one separately.

What should I do to fix it?

AndrewRH, 2019-09-17:
It should work.  You need a card that can do both input and output, and you also need to set the profile of the card for full-duplex mode.  You can either do this via the Video Setup application which lets you configure the default setup of the card.  Or you can go to the General menu in the Unity component and change the profile there.

Does that help?

AndrewRH, 2019-09-21:
@wsdfz Thanks for reporting this issue - we have replicated it and it is a bug in the latest version (a regression).  We're working on a fix now and so you should have a fix in the next few days.  I will let you know when there is a new version for you.

Thanks,

AndrewRH, 2019-09-22:
This should now be fixed - please upgrade to version 1.7.6 and see if that works.

Thanks,

closed 2019-09-23 by AndrewRH