RenderHeads/UnityPlugin-AVProDeckLink #10: Multiple Outputs

Panic77, 2019-12-10:
We are currently working on a project which controls an LED screen which takes multiple inputs. We want to target different output ports on the DeckLink 4k Extreme 12G card. Is this possible or do we need other hardware? 

How would we setup the Plugin to target different output ports?

AndrewRH, 2020-02-03:
Yes this should certainly be possible.  
You would need to configure the DeckLinkOutput component for having multiple outputs.  You can set the device profile here:
[image]

You can then have mutliple DeckLinkOutput components, each assigned to a different output index.
[image]

I hope this helps?  Or perhaps you found the solution yourself by now?

Panic77, 2020-02-04:
Hi Andrew,

Thanks for getting back to me regarding the multiple outputs.

I have another question regarding video input. I have created a unity app that takes a 1080p video input, which is displayed on one of our 4k monitors in sky studios.
We have seen that after around 24 hours the input texture (uGUIDisplay) seems to go black. I can fix this by simply reselecting the input resolution within the explorer which forces the texture to reinitialise.
Is this a known issue? Or is there something I can do to prevent this from happening?

Regards,

Pani

[image]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F5418126%2F73647797-2e96b200-4674-11ea-8dd1-78bf550eb1ff.png&data=02%7C01%7Cpanayis.chrysostomou%40sky.uk%7C1445b40c7efe456c071108d7a898434e%7C68b865d5cf184b2b82a4a4eddb9c5237%7C0%7C1%7C637163244253084781&sdata=DiR9ub%2B7Un3SPuEB1rhpna1nINwXmQofDEdCN1S%2B%2Fo8%3D&reserved=0>

[image]<https://eur01.safelinks.protection.outlook.com/?url=https%3A%2F%2Fuser-images.githubusercontent.com%2F5418126%2F73647890-5be36000-4674-11ea-8ac9-c56d8e76cee2.png&data=02%7C01%7Cpanayis.chrysostomou%40sky.uk%7C1445b40c7efe456c071108d7a898434e%7C68b865d5cf184b2b82a4a4eddb9c5237%7C0%7C1%7C637163244253094777&sdata=JEZ%2FRDFHwtWc3HlVZQOTi6iTgzIVpMOyxDAz4SDrz%2Fc%3D&reserved=0>

This email is from an external source. Please do not open attachments or click links from an unknown or suspicious origin. Phishing attempts can be reported by sending them to phishing@sky.uk<mailto:phishing@sky.uk> as attachments. Thank you
--------------------------------------------------------------------

Information in this email including any attachments may be privileged, confidential and is intended exclusively for the addressee. The views expressed may not be official policy, but the personal views of the originator. If you have received it in error, please notify the sender by return e-mail and delete it from your system. You should not reproduce, distribute, store, retransmit, use or disclose its contents to anyone. Please note we reserve the right to monitor all e-mail communication through our internal and external networks. SKY and the SKY marks are trademarks of Sky Limited and Sky International AG and are used under licence.

Sky UK Limited (Registration No. 2906991), Sky-In-Home Service Limited (Registration No. 2067075), Sky Subscribers Services Limited (Registration No. 2340150) and Sky CP Limited (Registration No. 9513259) are direct or indirect subsidiaries of Sky Limited (Registration No. 2247735). All of the companies mentioned in this paragraph are incorporated in England and Wales and share the same registered office at Grant Way, Isleworth, Middlesex TW7 5QD

AndrewRH, 2020-02-04:
Hmmm....that's strange.  We haven't seen that before...
We will try to replicate it here and get back to you.
My hunch is that it must be some sort of frame counter that's overflowing internally...

AndrewRH, 2021-10-04:
We haven't been able to reproduce this issue.

Is it still happening for you?

Otherwise we'll close this issue.

Thanks,

Panic77, 2021-10-04:
Production have not reported this issue happening again. Happy to close this.

closed 2021-10-04 by Panic77