RenderHeads/UnityPlugin-AVProDeckLink #13: Support for more than 2 audio input channels

mikerz, 2020-01-16:
we have use cases for 8, 16 or even 64 audio channels which we can’t pursue without this support. looking at online documentation, it looks like the input audio channels are hard coded to 2. we have tried more than 2 channels and it hasn’t worked, while 2 worked just fine

AndrewRH, 2020-02-03:
Great suggestion.  We will look into this.
How are you looking to use this amount of audio channels in Unity?  I believe Unity only supports up to 8 channels maximum itself...

mikerz, 2020-02-03:
I think it can actually support up to 12, based on what an audio contractor we're using let us know. Sending to audio sources would be good; as would a simple delegate method call that simply receives the buffer data

AndrewRH, 2020-02-04:
Do you want to manipulate the audio, or simply just send it back to the DeckLink for output?

mikerz, 2020-02-04:
In our case, we will be routing the data elsewhere in the system. We have our own native plugin, which renders the audio with Windows Sonic. 

Actually -- I took a look at the DecklinkInput class, and it looks like the code is already structured for however many audio channels the device supports? Based on some tests we did, only 2 channel audio worked over the SDI input while it came through fine in the default mediaexpress software as well as with having ffmpeg decode and play the audio.

Perhaps this is actually a bug..? Unless this was never fully implemented?

mikerz, 2020-02-14:
So, we've done some more testing and it looks like it isn't implemented. At 8 channels, the initialization process is successful, but within a minute it will crash Unity. At 16, 32 and 64, FMOD fails to initialize and unity will crash within a few moments of launching either the editor or a build. I believe it is a bug for the 8 channel option, and maybe greater than 8 should be disallowed by the plugin until there's a route forward?

pilzinho, 2024-01-15:
Maybe this can be revisited? Having the audio channels hardcoded to stereo is a bit limiting. We just started using the plugin but I'm sure that the requirement for more audio channels will come up at some point by a client.