RenderHeads/UnityPlugin-AVProDeckLink #16: 4K DCI 23.97 422 bit color resolution

gwxavi, 2020-02-24:
I have a machine with a **DeckLink 4K mini monitor** sending **4K DCI 23.97 at 10 bits 422** and a second machine capturing video using **DeckLink 4K mini recorder** same signal at 10 bits.
Unity plugin is used in the capture side.   All is working fine except by bit color resolution. When I run my app with AVProDeckLink plugin is set to 8 Bits. I tried to "force" to 10 bits but I get a black frame without signal detection. App works If I set with autodetection and it says is working at 8 bits. But it's not true, it's 10 bits.

I'm sure we sending 10 bits signal because we tested with a monitor in the middle doing bypass thought signal. Source video signal comes from Davinci Resolve Blackmagic setted with **4K DCI 23.97 at 10 bits 422** 
I'm playing with the trial version, maybe there is a color resolution restriction with the trial?

AndrewRH, 2020-02-24:
Hi,

Thanks for reporting this issue.  We will do some tests on our side.  You could also perhaps try running the Video Setup application (from BlackMagic) and see whether if you change the card settings between RGB/YUV whether that helps...

Thanks,

gwxavi, 2020-02-26:
Hi,

I reviewed our input settings using Video Setup App from BlackMagic but last version don't allow to change any input seetings.
here a screen capture of my video setup app
[image]

Let me know if you need more tests
thanks

AndrewRH, 2020-03-23:
@moz would be great if you could test this when you get a moment

RenderHeadsSte, 2020-08-07:
We have reproduced this and shall be looking to fix it in due course

AndrewRH, 2020-08-13:
We have a fix for this which should be included in the next version of the plugin (either tomorrow or early next week)

Thanks,

closed 2020-08-14 by AndrewRH

AndrewRH, 2020-08-14:
@gwxavi  this has been fixed now - thanks again for reporting it - the fix will be in the next version which will be released early next week.  Thanks,

AndrewRH, 2020-08-19:
Version 1.8.2 has now been released and fixes this issue.