RenderHeads/UnityPlugin-AVProDeckLink #17: DecklinkOutput uses high GPU load

AndrewRH, 2020-02-24:
When using DeckLinkOutput component, vsync is disabled to get the best accuracy for rendering the output frames.  This can be very wasteful and puts a high GPU and CPU load.  

It would be nice to improve this.  Either with the new method for rendering, or by adding some delays in there so that if it's running way too fast it will not max out the system.

HaruHappy, 2020-03-24:
how to set delay?

AndrewRH, 2020-03-24:
You could add a System.Threading.Thread.Sleep(8) into an Update() for example.

HaruHappy, 2020-03-24:
> You could add a System.Threading.Thread.Sleep(8) into an Update() for example.

i found a solution,set application.targetfram = 60     the cpu and gpu will going down

v2k, 2021-01-19:
Also noticing this is more of an issue after upgrading from 1.6.6 to 1.8.6; is setting the target frame rate the only work around?

AndrewRH, 2021-01-25:
We're looking into this now and will hopefully have more information by the end of the week

RenderHeadsMrT, 2021-05-28:
This issue may well have been linked to 'multi-output' being selected (if if only a single output was used) which would force Unity to run as quickly as possible. We've now automated this feature to only use multi-output mode if multiple outputs are in use and also limited Unity to a factor of the output rates instead of unbounded.

closed 2021-05-28 by RenderHeadsMrT