RenderHeads/UnityPlugin-AVProDeckLink #20: Add DirectX12 Support

MorrisRH, 2020-03-23:
Add support to the Windows plugin for the DirectX 12 API.

AndrewRH, 2020-09-17:
It's in the pipeline.   We have added DirectX12 support to 2 of our other plugins (AVPro Video and Movie Capture) and DeckLink is next.

Thanks,

Ludwig-Karl, 2021-02-16:
How does it look like? Is there a fixed date for the update? 

Thanks in advance

RenderHeadsMrT, 2021-02-17:
Hi @Ludwig-Karl,

As per Andrew's comment DeckLink is still next for DX12 support and should be starting within a week or two. We have to balance new features with support requests so unfortunately can't give an exact date but it will be in the region of weeks rather than months.

Cheers,

kamilsztuke, 2021-08-02:
Does AVPro DeckLink version 1.9.0 support DX12 ?

kahnivore, 2021-08-04:
Hi @kamilsztuke 

Not yet but it is being worked on currently. Will hopefully be released sometime in September.

closed 2022-07-28 by RichRH