RenderHeads/UnityPlugin-AVProDeckLink #21: Send capture to SDI.

Jotamaza, 2020-05-12:
**Is your feature request related to a problem? Please describe.**
I need to do a 180 capture and send it with audio in realtime using SDI or NDI.

**Describe the solution you'd like**
Be able to set the capture output to an SDI output using AVPro DeckLink would be perfect.

**Describe alternatives you've considered**
I've successfully sent the capture using NDI ([KlakNDI](https://github.com/keijiro/KlakNDI)), using the _finalTarget RenderTexture on CaptureFromCamera360.cs, but this plugin don't manage audio through NDI.

AndrewRH, 2020-05-14:
I will take a look at this today.  I can't see any reason why it shouldn't work.  We just just need to open up the API's a bit more.

AndrewRH, 2020-05-15:
Hi @Jotamaza 
I have attached here a new DeckLinkOutput.cs
The new version should allow you to specify a texture to output on DeckLink instead of having to use a camera for rendering.

You can call deckLinkOutput.SetSourceTexture(_finalTexture);
and make sure that the camera fields are set to null.

Let me know how this works for you.

Thanks,
[DeckLinkOutput.zip](https://github.com/RenderHeads/UnityPlugin-AVProDeckLink/files/4635029/DeckLinkOutput.zip)

Jotamaza, 2020-05-18:
Thanks Andrew! 

We are preparing a environment with DeckLink hardware to start the tests.
I will keep you up to date.

AndrewRH, 2020-05-18:
Okay great!  Can I close this issue or are you still testing it?

Thanks,

Jotamaza, 2020-05-19:
Hi Andrew! Here are my findings:

If I create a DeckLink output with the camera (as in the prefab), and then call the SetSourceTexture method, nothing changes in the output. The camera image is still streamed through the SDI port.

If I delete or deactivate the camera in the Output prefab there is also no change when calling SetSourceTexture.

The only way to get it working is start with the default camera, and then execute
sdiOutput.SetCamera(null);
sdiOutput.SetSourceTexture(captureTex);

Anyway, I got it working. Great!

Now, to complete my workflow, I just need to stop the AVPro Movie Capture output, as I don't need a video file or image sequence. As the stream could last for more than an hour, encoding and saving all that data is a performance consumption we can save. 
Any tip about where to stop the process without breaking something?

Thanks your support, Andrew!

Jotamaza, 2020-05-19:
Also the captured texture shows the post process effects, although the default camera behavior in  AVPro DeckLink don't support it.

AndrewRH, 2020-05-22:
> Also the captured texture shows the post process effects, although the default camera behavior in AVPro DeckLink don't support it.

Oh?  Could you explain more about this?   I don't think most post-process effects work for 360 though?

Jotamaza, 2020-05-25:
I have tested color grading and bloom from the Unity Post-processing Stack, and both where applied to the capture render texture.

AndrewRH, 2020-05-25:
I see :)  thanks!  Yes those two should work okay....Although bloom I would have thought might have edge seam artifacts....  But certainly color grading should work.

I will see if I can add 360/180 support as an example directly into the DeckLink package...