RenderHeads/UnityPlugin-AVProDeckLink #23: Questions about DecklinkOutput（1.7.8 ）

d0614, 2020-06-18:
NullReferenceException: Object reference not set to an instance of an object

RenderHeads.Media.AVProDeckLink.DeckLinkOutput.InitCaptureBlendResources (System.Int32 width, System.Int32 height) (at Assets/AVProDeckLink/Scripts/Components/DeckLinkOutput.cs:309)

RenderHeads.Media.AVProDeckLink.DeckLinkOutput.BeginDevice () (at Assets/AVProDeckLink/Scripts/Components/DeckLinkOutput.cs:757)

RenderHeads.Media.AVProDeckLink.DeckLink.Begin (System.Boolean search) (at Assets/AVProDeckLink/Scripts/Internal/DeckLink.cs:523)

RenderHeads.Media.AVProDeckLink.Demos.OutputDemo.OnGUI () (at Assets/AVProDeckLink/Demos/Scripts/OutputDemo.cs:128)

AndrewRH, 2020-06-18:
Thanks for reporting this.  We're investigating and will get back to you soon.

Thanks,

AndrewRH, 2020-06-18:
We have fixed the issue - version 1.7.9 is now available with the fix. Thanks for reporting this.  I will close this ticket now.

closed 2020-06-18 by AndrewRH