RenderHeads/UnityPlugin-AVProDeckLink #24: Frame dropping

RenderHeadsSte, 2020-06-24:
Issue brought over from keeping [#cTtVOv1NgA]

I made two identical cameras in Unity 2019.3.8.One for rendering scenes on PC and one for capturing card output.

The output picture has the problem of frame loss.

Video frame drop, unable to use.

I have recorded a video of the specific situation presented and sent it along with the email attachment.

RenderHeadsSte, 2020-06-24:
1) Could you verify that the frame drop is nothing linked to garbage collection?

2) We would be interested to see if the issue is linked to the output buffer not being filled fast enough. Could you run the output demo included in the asset package and send over the information displayed?

3) Have you tried running a build rather than in the editor?  The editor adds a lot of overhead.

d0614, 2020-06-29:
I'm not sure if this problem is related to GC because I'm new to GC and don't know much about it.But the same thing happens when I run the output demo in the asset package (05_OutputDemo).After the project is build, this issue is not resolved.

Describe my problem accurately again:
When the output camera stays still, the output looks fine.But when the output camera starts to move, the output image will shake.I used the Iphone's slow motion mode to shoot a comparison between the unity running screen and the output screen of the capture card, and found the frame drop problem.

d0614, 2020-06-29:
> 1. Could you verify that the frame drop is nothing linked to garbage collection?
> 2. We would be interested to see if the issue is linked to the output buffer not being filled fast enough. Could you [...]

I'm not sure if this problem is related to GC because I'm new to GC and don't know much about it.But the same thing happens when I run the output demo in the asset package (05_OutputDemo).After the project is build, this issue is not resolved.

Describe my problem accurately again:
When the output camera stays still, the output looks fine.But when the output camera starts to move, the output image will shake.I used the Iphone's slow motion mode to shoot a comparison between the unity running screen and the output screen of the capture card, and found the frame drop problem.

AndrewRH, 2020-06-30:
Hi @d0614 

Thanks for the feedback, it will help us to figure out this issue.

Please could you run the 05_OutputDemo scene and when it's outputting there are some stats printed to the top-right corner of the screen.   These show the buffer health.  Let me know what those numbers show as they could indicate that the output buffer is either filling up too quickly, or not fast enough, both of which would result in frame drops.  

Also, which exact mode are you outputting?  (resolution, interlaced/progressive and frame rate)

Thanks,

d0614, 2020-06-30:
你好

对不起，我的英语水平不够好，无法正确理解您。

我希望这次我能正确理解。随附件一起发送了运行05_OutputDemo的视频。

我没有明确的输出模式，主要是1080p / 60.

感谢您耐心的回复

------------------&nbsp;原始邮件&nbsp;------------------
发件人:&nbsp;"Andrew"<notifications@github.com&gt;;
发送时间:&nbsp;2020年6月30日(星期二) 晚上6:18
收件人:&nbsp;"RenderHeads/UnityPlugin-AVProDeckLink"<UnityPlugin-AVProDeckLink@noreply.github.com&gt;;
抄送： &nbsp;“ 353791975” <353791975@qq.com&gt;;“提及” <mention@noreply.github.com&gt;;
主题:&nbsp;Re: [RenderHeads/UnityPlugin-AVProDeckLink] Frame dropping (#24)

 
嗨@ d0614
 
感谢您的反馈，它将帮助我们找出此问题。
 
请运行05_OutputDemo场景，输出时，在屏幕右上角会打印一些统计信息。这些显示了缓冲区的运行状况。让我知道这些数字所显示的含义，因为它们可能表明输出缓冲区填充得太快或不够快，这两者都会导致帧丢失。
 
另外，您要输出哪种确切模式？（分辨率，隔行/逐行和帧速率）
 
谢谢，
 
- 
您收到此邮件是因为有人提到您。
直接回复此电子邮件，在GitHub上查看，或取消订阅。

d0614, 2020-06-30:
你好

对不起，我的英语水平不够好，无法正确理解您。

我希望这次我能正确理解。随附件一起发送了运行05_OutputDemo的视频。

我没有明确的输出模式，主要是1080p / 60.

感谢您耐心的回复

------------------ 原始邮件 ------------------
发件人:&nbsp;"Andrew"<notifications@github.com&gt;;
发送时间:&nbsp;2020年6月30日(星期二) 晚上6:18
收件人:&nbsp;"RenderHeads/UnityPlugin-AVProDeckLink"<UnityPlugin-AVProDeckLink@noreply.github.com&gt;;
抄送： &nbsp;“ 353791975” <353791975@qq.com&gt;;“提及” <mention@noreply.github.com&gt;;
主题:&nbsp;Re: [RenderHeads/UnityPlugin-AVProDeckLink] Frame dropping (#24)

 
嗨@ d0614
 
感谢您的反馈，它将帮助我们找出此问题。
 
请运行05_OutputDemo场景，输出时，在屏幕右上角会打印一些统计信息。这些显示了缓冲区的运行状况。让我知道这些数字所显示的含义，因为它们可能表明输出缓冲区填充得太快或不够快，这两者都会导致帧丢失。
 
另外，您要输出哪种确切模式？（分辨率，隔行/逐行和帧速率）
 
谢谢，
  
- 
您收到此邮件是因为有人提到您。
直接回复此电子邮件，在GitHub上查看，或取消订阅。

从QQ邮箱发来的超大附件

bandicam 2020-06-30 18-24-28-875.avi (91.61M, 2020年07月30日 18:31 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?k=6e3931302d0a25cbd05627254330511e04005203510856551a5f5303031d5709070c1c085754561c0300520404035002075b5705651463535657555906510e11050903004800551c040911015d1d51051a0b091d5d07561f564f583058&t=exs_ftn_download&code=7910e0c1

AndrewRH, 2020-06-30:
Thanks @d0614 

I saw the video.  Could you send the original slow-motion capture video?

If you disable that debug output GUI, can you still see frame-drops?   That debug output GUI generates garbage, which can make the GC trigger and cause a frame drop.  Perhaps you could check the profiler as well?

Also what GPU are you running on?  Have you checked the Control Panel to make sure it's set to Maximum Performance?

Thanks,

d0614, 2020-06-30:
hi,

I'm really sorry that there are problems in the first two emails.

The video of the 05_OutputDemo scenario running has been relaxed with the attachment.

Thank you for your reply.

 

 
 
 
 
 

从QQ邮箱发来的超大附件

bandicam 2020-06-30 18-24-28-875.avi (91.61M, 2020年07月30日 18:37 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?k=3f3432382c5c71ca815b242d4266051f07070501025f0f034b00540d5d4b0300570d1f000004021d5107505a5c5f56005f06075b64423752075a565107075a10540400084956011d550412095c4b05044b060a155c51021e07425b3859&t=exs_ftn_download&code=f428df70

d0614, 2020-06-30:
This is a slow motion video that was shot earlier.

Since I saw the questions you answered on Github before, I have set the GPU as the Maximum Performance.

I used GTX1080.

I'm not sure if it's a GC problem.When I create a scene with a few simple models and a script that outputs the number of GC, the log information shows that the number of GCS increases.At this point the output will drop frames.
If GC is causing the frame drop problem, how can I solve it?There are no other scripts in the scenario.

 

 
 
 
 
 

从QQ邮箱发来的超大附件

IMG_2005.SVP.MP4 (115.82M, 2020年07月30日 18:49 到期)进入下载页面：http://mail.qq.com/cgi-bin/ftnExs_download?t=exs_ftn_download&k=73323839ddb3f79ad15d2e2c1f64074d57570e0c0b5602571b025c585a490157015315000d00514f0f04000d09065607050b0d0e3974352b7b75670b0954004c65646817743401620b&code=62899d5b

AndrewRH, 2020-07-01:
Oh,

I meant the original slow-mo video file - since the one you sent us was a recording of that.

Thanks,

RenderHeadsMrT, 2021-05-28:
We've made some optimisations to the output pipeline feeding textures from Unity to DeckLink which have given noticeable performance improvements. On top of this we've added an output health status that gives feedback as to how well data is buffered (closer to zero indicates frames aren't being fed through fast enough from Unity).

If you're still seeing issues it's worth looking at the output health status - if this is not around green/100% then it would indicate frames aren't being fed to DeckLink fast enough for the desired output rate. This could be caused by a host of issues such as an overly complex scene (geometry detail, post-processing etc..) or stalls in Unity (such as GC, editor/UI rendering etc..).

closed 2021-05-28 by RenderHeadsMrT