RenderHeads/UnityPlugin-AVProDeckLink #27: GenLock offset is always overwrite to zero in player mode

buresu, 2020-08-01:
The GenLock offset is always overwrite to zero in player mode.
This problem doesn't occur in editor mode.

Step to reproduce:

1. Open DeckLinkOutput component and change Genlock Pixel Offset.
2. Run the game on editor.
3. Open DesktopVideoSetup and check the Refernce Input offset. <- No problem 
4. Build and Run the game.
5. Open DesktopVideoSetup and check the Refernce Input offset. <- Wrong  

Environments:
Unity 2019.4.6
DeckLink 8K Pro
DeckLink 11.5.1
AVPro DeckLink 1.7.6

I hope a option to not change the genlock offset.

AndrewRH, 2020-08-11:
Thanks for reporting this issue.

Are you making 32-bit or 64-bit builds?

Thanks,

buresu, 2020-08-11:
Thank you for the reply.

I think it was probably 64bit.
My Platform is windows 10 1909 x64.

AndrewRH, 2020-08-11:
> Thank you for the reply.
> 
> I think it was probably 64bit.
> My Platform is windows 10 1909 x64.

Thanks!  But could you check your build settings?  I suspect it may be set to build to 32-bit EXE, which could be causing problems....   64-bit build may be fine.

Please let me know as it is a clue to solving this bug.

Thanks,

buresu, 2020-08-11:
Sure, I'll check tomorrow.

AndrewRH, 2020-08-12:
We have replicated the bug and are working on a fix.

buresu, 2020-08-12:
Thank you!

> Thanks! But could you check your build settings? I suspect it may be set to build to 32-bit EXE, which could be causing problems.... 64-bit build may be fine.

Just in case, there was no difference between 32bit and 64bit build in my environment.

AndrewRH, 2020-08-13:
This has been fixed and will be in the next release of the plugin (which should be either tomorrow, or early next week)

I will close this issue - but thanks again for reporting it and for your patience while we fixed it.

Thanks,

closed 2020-08-13 by AndrewRH

buresu, 2020-08-13:
Thank you so much :)
I'm looking forward to the next release.

AndrewRH, 2020-08-19:
Hi
The release is now available on the asset store, and trial version is here:
https://github.com/RenderHeads/UnityPlugin-AVProDeckLink/releases/tag/1.8.2

Thanks again for reporting this issue.

buresu, 2020-08-19:
I am glad to hear that news!
Thank you as well :)