RenderHeads/UnityPlugin-AVProDeckLink #28: Unity packaged project, SDI GPU occupancy 100% problem?

gy751, 2020-08-06:
Unity packaged project, SDI INPUT/output GPU occupancy rate of 100.
Can this problem be solved?How do I do that?

AndrewRH, 2020-08-06:
This should only happen on Output I believe.  It's because we disable vsync, so the GPU runs as fast as possible which allows us to better sync with the DeckLink card so we make sure that no frames are dropped.  You could always add a Sleep in there to give the GPU a bit of a rest..  But it's already on our list to look into here: #17 

Thanks,

RenderHeadsMrT, 2021-05-28:
This issue may well have been linked to 'multi-output' being selected (if if only a single output was used) which would force Unity to run as quickly as possible. We've now automated this feature to only use multi-output mode if multiple outputs are in use and also limited Unity to a factor of the output rates instead of unbounded.

closed 2021-05-28 by RenderHeadsMrT