RenderHeads/UnityPlugin-AVProDeckLink #29: Image Quality Issues

thegladscientist, 2020-09-28:
**Describe the bug**
The image quality of the video through the plugin is fuzzy compared to other programs (OBS, TouchDesigner, Resolume, and even Keijiro's Unity plugin).

**Your Setup (please complete the following information):**
 - Unity version: 2019.3.7, 2019.4.9, 2020.1.0
 - AVPro DeckLink version: 1.8.5, previous version before
 - Operating system version: Windows 10
 - Hardware model: Decklink Intensity 4k, Decklink Mini Recorder 4k, Decklink Duo
 - Cables used: HDMI, SDI
 - Desktop Video version: 11.5.1, previous version before
 - Video mode (resolution, frame-rate, pixel-format): 1080p, 60fps

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...' any Unity project with Decklink plugin. 
2. Click on '....' plugin and then select proper input. Press play in the editor. 
3. _Adjust settings in the plugin explorer...they do not create any effect.

**Screenshots**
You can see here the quality difference between Keijiro's plugin and the avPro plugin. Look at the rivets on the case and you can see the detail is lost. These are compared in the same project with the same settings and hardware, just the capture plugin enabled/disabled. It feels subtle at first, but this difference has meant a lot of extra work for my team, and a loss of clients.

AvPro:
[image]

Keijiro:
[image]

AndrewRH, 2020-09-28:
Thanks for reporting this issue.

Could you tell us exactly which mode you're using (eg RGB-10 bit or YUV-8 bit) and whether your signal is interlaced or not.

Also - does it make any difference if the automatic deinterlace option is disabled?
[image]

With these pieces of information we should then be able to replicate the issue on our side and fix the problem.

Thanks,

AndrewRH, 2020-09-28:
OH and also turn off the AutoDeinterlace option too just to be sure..
[image]

Does that change anything?

I'm also curious to know how you're grabbing these screenshots - how are they being displayed?  I'm wondering whether there is some tiny amount of scaling.  You can also click on the Save PNG button to save a PNG to the project folder
[image]

The PNG should be the best way to evaluate the quality.  If there is a scaling issue during the display of this texture then it coudl be causing blurriness.

Thanks,

thegladscientist, 2020-09-29:
Thanks so much for the quick reply! I have tried every mode and color space option available, but when opening the explorer in play mode it seems to reset it automatically to 4:2:2 YUV. For reference to my camera, I am using the BlackMagic Pocket Cinema 4k. 

Re the screenshots...I just used snipping tool with the same scale in the Unity editor's viewer. Since the other plugin does not have the save PNG function directly, this was the easiest way to show a still of the same scale and exact shot. As a note, I observed the quality difference before taking the screenshots, so it is apparent in a "A/B Test".

I just tried to do the deinterlace method now (it was on before), and here are the results (no observable improvement unfortunately):

Deinterlace BLEND, Auto Deinterlace ON:

[image]

Deinterlace OFF, Auto Deinterlace ON: 
[image]

Deinterlace OFF, Auto Deinterlace OFF:
[image]

AndrewRH, 2020-09-29:
Okay thanks for more information.

I think the quality is actually the same, but it's the way the texture is being displayed that is causing it to look worse - due to a combination of a tiny bit of scaling, some filter, or both.  If you're viewing it in the editor window, then it will probably be scaled - so unless the same display method is being used, it's not usually a fair comparison.

We had a similar email from a company called Cosmic Lab Japan - was this related to your issue?

We will look into this further and get back to you.

thegladscientist, 2020-09-30:
Yes, that is my company as well, didn't realize a coworker had emailed already. 

We have used this plugin over multiple projects in the past, and have received feedback from clients about video quality not being the same as the raw video footage we capture directly through the camera. Similarly to the first comparison images, the image through the AvPro plugin loses sharpness in places of high detail. 

All footage is captured in a build (i.e. no scaling) and when compared to raw footage coming in through an SDI splitter before the capture card, is noticeably different. It is requiring us to recomposite 3D content in post as opposed to keeping the original footage with 3D composited, so this is a high priority issue for us, as it could save many hours of work.

As you mentioned, the quality is the same in the second comparison, where interlacing is toggled on or off, all within the AvPro plugin. 

Do you have any other ideas for us to try while your team is looking into it? We have a recording this weekend so would like to try to test what we can before then. :)

AndrewRH, 2020-09-30:
Is there any chance you could send us some better test images?  Ideally we would have:

1) The input set up with something like laptop/pc desktop displayed on it - with a background containing a detailed test pattern, or the camera pointed as a brighter scene with more details.

2) Get that image in Media Express and save it to TGA

3) Get that image from AVPro DeckLink via the Save PNG button

Then we can really compare exactly what is happening.

I have done the above in our test rig here and the resulting image is the same as that from Media Express, which is why I'm not sure how to help further without more clues.

Thanks,

AndrewRH, 2020-09-30:
Also - could you try updating the drivers to the latest version of Desktop Video (11.6)?

thegladscientist, 2020-09-30:
Thanks for all the responses!

Attached is the test requested. You can see the difference in the input from the Unity plugin vs. the BlackMagic Express. 

**AvPro Plugin:**

[image]

[image]
[image]

**BlackMagic Media Express:**
https://www.dropbox.com/sh/npm57sd4ia1youh/AAD5T-FX37dDh9PKigj75w_2a?dl=0

thegladscientist, 2020-09-30:
additionally, here are my settings in the DeckLink plugin in Unity (I created a new empty GameObject) 

[image]
[image]

AndrewRH, 2020-09-30:
Thanks very much for those images - very helpful.

We've released a new version of the plugin which fixes a number of issues - please update to 1.8.6 and let me know if there is any improvement.

If there is still no improvement please try:

1) switching the colorspace from Linear to Gamma in Player Settings
2) see if you can force the mode to 10-bit rgb for input
3) let us know which demo scene you're seeing the fuzzy visual issue in - make sure that in any scene that the viewport is set to 1920x1080 with no scaling, run the game view maximised on a monitor > 1080p etc.
4) try other demo scenes - eg the input explorer scene you can click on the preview to make it fullscreen
5) supply us again with any helpful screenshots / saved pngs that you can

Thanks,

thegladscientist, 2020-10-01:
Hi Andrew,

Thank you for this update. 

I have done a series of tests with the new plugin and it looks much better overall. However, we still need to do some focus related tests coming through an actual camera to be certain. There is a slight difference in color (darker image in the plugin than in BME). I tried various settings shown below to try to match the color of the BlackMagic Express but it always came out darker. Please see the following images for that finding:

Linear, Default Settings
[image]

Gamma, Default Settings
[image]

Gamma 10bit RGB
[image]

Gamma, DeInterlace OFF, 10 bit RGB
[image]

Gamma, DeInterlace OFF, 10bit, Rec2020 Color Space
[image]

You can compare this with the same corresponding image from before, as I used it as a constant in the testing: https://www.dropbox.com/sh/npm57sd4ia1youh/AAD5T-FX37dDh9PKigj75w_2a?dl=0

Please let me know if you have any ideas here, and I will report back with a detail test through the camera tomorrow. 

Thanks again!

AndrewRH, 2020-10-01:
Thanks for testing - and if you could try upgrading to the latest DeckLink Video (11.6) as well please for your camera tests.

Thanks,

thegladscientist, 2020-10-01:
Sure, we’re already on 11.6 so no worries there. Will share the tests after today’s results. Thanks again.

cosmic1, 2020-10-04:
Hey Andrew, thanks for helping us out with this issue. We found out that the latest plugin update seemed to do the trick and the exported PNG now matches the black magic express TGAs. After that we found an additional issue with Unity HDRP applying the default post processing settings to the quad we use to display the camera output (even though we set the render camera to ignore post processing volumes). To fix the final issue we had to change the render pass/queue from "default" to "after post-processing".

AndrewRH, 2020-10-04:
Very glad to hear that the issue has been resolved.  Thanks for your help with this one, and do let us know if you have any more issues in the future.  I will close this thread now.
Thanks,

closed 2020-10-04 by AndrewRH