RenderHeads/UnityPlugin-AVProDeckLink #30: Input from Blackmagic Card

battonm39, 2020-10-07:
Using the Input demo on Decklink plugin (trail version) an  error pop once I played the demo. Below is the error msg. I have not changed any settings from default. My camera input is through a blackmagic capture card (Verified that the card is working properly). Camera is a Blackmagic 4K SDI out and settings are 1080i 60f. 

Just a little BG on what I'm trying to accomplish. Need to import three different video sources into Unity from a production truck.   
   
[AVProDeckLink] Desired format not found - the resolution or pixel format may not be supported

Can you point me in the right direction to start on the problem?

AndrewRH, 2020-10-08:
That should just be a warning.  After that error is anything else logged?  

Perhaps try using the options to force the mode to start with:
[image]

Does that help?

Thanks,

AndrewRH, 2020-10-12:
Any luck?  I'd love to be able to get the bottom of this one.

battonm39, 2020-10-12:
Hi Andrew, I tried your suggestions and didn't have any luck. I did download and try the AvPro LiveCam and it pop right up when I played the project. Very strange. I went back through and did get the Decklink to output my main camera. I once again tried to capture a live feed from a camera and still nothing (from AVPro Decklink). Later today I will capture screen grabs of the inspector. See if we can't get the bottom of this. Thank you for your help. Love the plugin!

Marcus

AndrewRH, 2020-10-12:
Hi Marcus,

Do you have the Desktop Video 11.6 installed?

battonm39, 2020-10-13:
Hey Andrew, I have figured out the problem.  Problem was with my script calling the wrong shader. Thank you for all your help!!!

AndrewRH, 2020-10-13:
Glad you solved it :)

closed 2020-10-13 by AndrewRH