RenderHeads/UnityPlugin-AVProDeckLink #31: Problem with Post Processing - Lose the input video from blackmagic card

battonm39, 2020-10-28:
Hi,

Working with AVPro Decklink plugin and everything was working perfectly. But when I start to use post processing (FXAA) to the project and build my project. I suddenly lose my two input video feeds from the Blackmagic card. Any ideas and what step I am missing or does the plugin not like to use post processing? Thanks 

UNITY V2019.3.14f1

Marcus

AndrewRH, 2020-10-28:
Hi Marcus,

Thanks for reporting this.

Could you tell me which version of AVPro DeckLink you're using and also which version of the post-processing?

Are you using either HDRP or URP?

What exactly do you mean by 'lose my two input video feeds'?  Is the hardware no longer detected or is just a black frame returned?

Thanks,

battonm39, 2020-10-29:
I am currently using Post Processing V2.3.0 also the version of AvPro is 1.8.6. I am not using HDRP or URP just the built in. 

Yes sorry, the two feeds just display black instead of the two camera feeds coming in through the blackmagic card. I shut off post processing and built the project again and the video feeds returned. 

Marcus

AndrewRH, 2020-10-30:
Hi,

I tested the plugin with Post Processing 2.3.0 with Unity 2019.4.13 (slightly newer than your version), and it worked fine.  I used a couple of our demo scenes, and added a Layer and Volume post processing component to them and a bloom filter.  Everything worked fine in the test.

Could you explain exactly how your scene is set up?  Perhaps include a screenshot or tell us how to reproduce it from one of our demo scenes.

Thanks,

battonm39, 2020-11-02:
So I have taken screen shots of my basic setup. I have two planes in a framed box. I have attached each plane with the Decklink input script. The camera has Cinemachine installed and running a dolly back and forth.

Thank you  

[image]
[image]
[image]
[image]
[image]
[image]

Chris-RH, 2022-05-13:
CNR

closed 2022-05-13 by Chris-RH