RenderHeads/UnityPlugin-AVProDeckLink #33: Black Texture with MipMaps enabled

D10scur0, 2020-12-24:
Hi there,

trying out the trial version of the plugin, when mipmaps is enabled on the created render texture, the input image is not coming thorught.

**Setup**
 - Unity version: 2019.3.15f1;
 - AVPro DeckLink version: 1.8.6t;
 - Operating system version: win 10;
 - Hardware model: Blackmagic 4k hdmi;
 - Video mode (resolution, frame-rate, pixel-format): 4k UHD, 60fps, 10bit 4:2:2 YUV;

**To Reproduce**
Steps to reproduce the behavior:
0. Setup a scene with a DecklinkInput and ApplyToMaterial prefab;
1. Go to FormatConverted.cs;
2. Locate RecreateConvertedTexture() method;
3. Add to the texture configs texture.useMipMap = true; and  texture.autoGenerateMips = true; 
4. Go in Play and observe the Issue: the texture is black.

On a second note not related to the issue; To display 4 concurrent input, is the right procedure to create 4 instances of DeckLinkInput(each with its own index) and 4 ApplyToMaterial?

Thanks

D10scur0, 2021-01-04:
As a workaround, at the moment, I'm blitting the OutputTexture into a Render Texture applied directly to the material. Pls let me know if there are updates, or better workarounds.

Ty

AndrewRH, 2021-01-05:
Yes, your blitting workaround is the only way to do this.  Changing the texture settings as you did previous will not work as the external texture is not a RenderTexture.  Improving native mip-map support is something that is on our roadmap.   Effectively we would just have to replicate what you're doing, but internally.

I will change this to an enhancement and leave this open until we have implemented the solution.

Thanks for bringing this to our attention - mipmap's are certainly very useful in many cases.

D10scur0, 2021-01-05:
Hi Andrew, 

thanks for the reply. Will look forward the enhancement.