RenderHeads/UnityPlugin-AVProDeckLink #34: Flipping X, Y influences multiple inputs when the sources come from the same hardware

D10scur0, 2021-01-04:
**Describe the bug**

Flipping X or Y when using the plugin with a multiple input card will influences different inputs. See attached video for a clear overview of the issue.

Apparently the issue is present when the sources come from the same hardware, but still on different Inputs/Outputs. 

To be more explanatory in the setup: As a source we connected 2 input (0 and 1 on the Decklink card) from 2 different outputs from the graphic card of a computer, and the other 2 (2 and 3) from another.

**Your Setup (please complete the following information):**
 - Unity version: 2019.3.15f1- HDRP
 - AVPro DeckLink version: 1.8.6t
 - Operating system version: Win 10
 - Hardware model: DeckLink Quad HDMI Recorder
 - Desktop Video version: 11.7
 - Video mode (resolution, frame-rate, pixel-format): 

**To Reproduce**
Steps to reproduce the behavior:
1. Create a new empty project in unity;
2. Import AvPro Decklink Trial Package;
3. Open 04_InputExplorerDemo scene;
4. Connect 2 Outputs from a Graphic card of one computer to 2 Inputs in a BlackMagic Acquisition card;
5. Play and try to flip one of the inputs;
6. Observe the issue as both the inputs get flipped;

**Videos**

https://user-images.githubusercontent.com/59603519/103538299-3f5c3300-4e96-11eb-8b55-7247a0b97407.mp4

AndrewRH, 2021-01-04:
Thanks for reporting this - we are investigating.

D10scur0, 2021-01-07:
Hi Andrew,

if it can be of any help, I'm raw guessing, the culprit could be in the conversion material (See picture) shared by the inputs.

[image]

Disregard this comment if totally off track.

AndrewRH, 2022-05-13:
You are correct!  We will address fix (and sorry it's taken so long)

RichRH, 2023-12-05:
Closing as fixed in https://github.com/RenderHeads/UnityPlugin-AVProDeckLink/releases/tag/1.9.4.

Cheers,

closed 2023-12-05 by RichRH