RenderHeads/UnityPlugin-AVProDeckLink #39: Tally Support

cwnuk-lfl, 2021-01-25:
It would be really helpful for us to have support for reading SDI tally information on outgoing connections, and setting tally information for incoming connections.  More information on page 201 of this Blackmagic manual: https://documents.blackmagicdesign.com/UserManuals/ATEM_Production_Studio_Switchers_Manual.pdf?_v=1594364410000

AndrewRH, 2021-01-26:
Thanks for the suggestion.

So that would send and receive tally by embedded it into the VANC?

Thanks,

cwnuk-lfl, 2021-01-26:
Correct, so when using SDI out, Unity would have an understanding of whether or not that feed is previewed or live on a production switcher, and when using SDI in, Unity could illuminate the tally light on the camera when it is using its feed.