RenderHeads/UnityPlugin-AVProDeckLink #40: Question: AVProDeckLink only works with BlackMagic?

EricBatlle, 2021-02-04:
I want to know which graphic cards and brands are compatible with this plugin apart from BlackMagic cards to renderize through SDI, or if it's necessary a conventional GPU card.

Thanks

kahnivore, 2021-02-04:
Hey @EricBatlle 
AVPro DeckLink is only for BlackMagic DeckLink (and some Intensity) capture/playback hardware (all of these products basically under 'capture and playback': [https://www.blackmagicdesign.com/products](url))
Some sort of GPU is generally needed for hardware accelerated rendering, but the requirements of that will depends on 
1) how complex visuals the visuals are being rendered
2) the resolution you want to work with for the Decklink input/output frames
We offer free trials of our products here if you'd like to test something [http://renderheads.com/downloads/](url)

EricBatlle, 2021-02-04:
Thanks @kahnivore !

closed 2021-02-04 by EricBatlle