RenderHeads/UnityPlugin-AVProDeckLink #41: Desired device not found

RiMichiA, 2021-02-16:
Hey, I downloaded the trial version today.
But unfortunately it does not work. The plugin does not find my decklink card. In Device Manager the card is recognized and in Blackmagic Desktop Video Setup too. Can someone please help me. Probably it is some simple error on my part.

[image]
[image]

RenderHeadsMrT, 2021-02-17:
Hi @RiMichiA,

Unity version, flags etc.. look good - just to double-check, you're using Desktop Video 11.6.0 or greater?

The "[AVProDeckLink] Desired device not found" warning you're seeing is more suggestive of a mode than a device issue - if the device itself wasn't found then we'd also expect to see "[AVProDeckLink] Unable to find any DeckLink Devices, It is possible that your Desktop Video is out of date. Please update to version 11.6.0" logged.

You could try setting the '_logDeviceEnumeration' flag to true in 'DeckLinkManager.cs' (or adding the 'Deck Link Manager' component and ticking the 'Log Device Enumeration' flag but this may throw singleton errors and should only be temporary for testing) - this should provide some more verbose details that can hopefully help narrow things down.

Cheers,

RiMichiA, 2021-02-23:
Hi @RenderHeadsMrT,

thank you for your answer. You were right, I didn't have the right version of Desktop Video.... Unfortunately I can't test it in the studio at the moment, but I guess it will work with the new version.

Thanks a lot!

RenderHeadsMrT, 2021-02-23:
Thanks for confirming @RiMichiA - I'll close this ticket now but please do repoen it if you encounter further issues.

closed 2021-02-23 by RenderHeadsMrT