RenderHeads/UnityPlugin-AVProDeckLink #44: DecklinkInput OnAudioFilterRead() conflicts with NDI AudioFilter

D10scur0, 2021-06-24:
**Describe the bug**
Y-hello, I'm integrating AVProDecklink(trial version for now) and KlakNDI plugin by Keijiro. For some unknown reason when the NDI audio is playing and the DecklinkInput is not on mute, the buffers seems to collide. I'm trying to wrap my head around but no luck.

**Your Setup (please complete the following information):**
 - Unity version: 2019.3.15f1
 - AVPro DeckLink version: 1.8.6t
 - Operating system version: win10 x64
 - Hardware model: Decklink Quad HDMI
 - Cables used: HDMI
 - Desktop Video version: 11.7
 - Video mode (resolution, frame-rate, pixel-format): Not related

**To Reproduce**
Steps to reproduce the behavior:
1. Implement this fork of KlakNDI "https://github.com/benwager/KlakNDI/tree/send_recv_metadata-updates"
2. Try to Mute/Unmute a DecklinkInput (even if is not receiving anything)
3. Observe the issue as the NDI audio starts to crackle and falls behind
4. Observe the DecklinkInput audio filter getting influenced

**Videos**
Video/Audio of the issue
https://vimeo.com/567093639

D10scur0, 2021-06-28:
[Update]

Fiddling about I discover that the issue is present also for regular AudioSource playing an imported clip, but only when there isn't any incoming input in the DeckLinkInput. 

When the decklink is receiving the issue is not present but still muting/unmuting causes the audio to crackle for a bit.

Anyway I could work around this issue? Any suggestion would be much appreciated.

Thank you

D10scur0, 2021-07-14:
Hi @AndrewRH,

by any chance did you had some time to look into this?

Is there any possibility to take a peek at what
`[DllImport("AVProDeckLink")]
		public static extern void GetAudioBuffer(int device, float[] buffer, int size, int channels, float volume);`
is doing? 

Hard guess is that if there isn't any audio input coming from the device, the methods will try to get any other buffer?

Ty

AndrewRH, 2022-05-13:
This looks like a real issue.  It could be purely performance based if the audio code is all in C#, so we could optimise that.  But there may also be some sort of logic bug with the buffering.  We'll need to investigate.

Thanks,

RichRH, 2023-12-06:
Closing this and merging with #55 as I believe it's the same issue - see my comment re: audio thread stalls as this would adversly affect anything else (e.g. NDI) using Unity's audio thread.

Cheers,

closed 2023-12-06 by RichRH