RenderHeads/UnityPlugin-AVProDeckLink #46: Output fails with: Unable to start device Intensity Pro 4K, it might be currently busy

genereddick, 2021-07-28:
**Describe the bug**
A clear and concise description of what the bug is.

In the latest version 1.90 I am unable to get an Output feed from a card when Input is running. The previous version of the software with all other settings the same handles this fine.

The issue appears in this code from the current version:

From Declink.cs
`
public void Begin(bool search = false) {
...
if (_device != null)
{
	bool valid = IsInput() ? _device.CanInput() : _device.CanOutput();
       if (valid)...
       else {
	     Debug.LogWarning("[AVProDeckLink] Unable to start device " + _device.Name + ", it might be currently busy");
	   _device = null;
	}
`

CanOuput Which calls into device.cs
`
public bool CanOutput()
		{
			bool result = SupportsOutput && !IsOutputBusy && !IsInputBusy;
			/*if (result)
			{
				if (_fullDuplexSupported)
				{
					result = !IsStreamingOutput;
				}
				else
				{
					result = !IsStreaming;
				}
			}*/
			return result;
		}`

Here  IsInputBusy is returning true causing the Output to fail. If I hacking this to return true it works. The previous version of this code (pre 1.9.0) also works.

`public bool CanInput()
		{
			if (_fullDuplexSupported)
			{
				return !IsStreamingInput;
			}
			else
			{
				return !IsStreaming;
			}
		}`

**Your Setup (please complete the following information):**
 - Unity version: 2020.3.14f1
 - AVPro DeckLink version: 1.9.0
 - Operating system version: Windows 10
 - Hardware model: Intensity Pro 4K or Ultrastudio
 - Cables used: HDMI
 - Desktop Video version: 11.7
 - Video mode (resolution, frame-rate, pixel-format): 1080p, 24fps
 - 

**To Reproduce**
Steps to reproduce the behavior:
Download latest version, attempt to run InputOutputDemo scene, using Decklink Intensity Pro 4K or Ultrastudio. Check that Input and Output both work.

**Logs**
[AVProDeckLink] Unable to start device Intensity Pro 4K, it might be currently busy
UnityEngine.Debug:LogWarning (object)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin (bool) (at Assets/AVProDeckLink/Scripts/Internal/DeckLink.cs:565)
RenderHeads.Media.AVProDeckLink.Demos.InputModifyOutputDemo:StartOutput (RenderHeads.Media.AVProDeckLink.DeviceMode) (at Assets/AVProDeckLink/Demos/Scripts/InputModifyOutputDemo.cs:151)
RenderHeads.Media.AVProDeckLink.Demos.InputModifyOutputDemo:OnGUI () (at Assets/AVProDeckLink/Demos/Scripts/InputModifyOutputDemo.cs:364)

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Videos**
If applicable, add a copy of your video or the URL

RenderHeadsMrT, 2021-09-22:
Hi @genereddick - just wanted to check if the above fix worked for you?

Cheers,
Richard

genereddick, 2021-09-22:
Yes, the fix worked. Thanks!

closed 2021-09-23 by AndrewRH