RenderHeads/UnityPlugin-AVProDeckLink #48: HDR Rec 2020 rainbow striping on output

thomasfdm, 2021-08-17:
**Describe the bug**
Attempting to get HDR our using Rec 2020 however output has odd rainbow striping on output. Tried every HDR setting we can find within Unity and still makes no difference. Is there a guide on how to achieve HDR video out?

Monitor output looks as expected though

**Your Setup (please complete the following information):**
 - Unity version: 2019.4.28f1 and 2020.3.14f1
 - AVPro DeckLink version:  1.6.6 and 1.9.0
 - Operating system version: Windows 10 
 - Hardware model: DeckLink 4K Extreme 12G
 - Cables used: BNC SDI
 - Desktop Video version: 12.1
 - Video mode (resolution, frame-rate, pixel-format): 1080p 50 10bitYUV and 8bit YUV

**Logs**
N/A

**Screenshots**
Unable to share due to scene's content (however an email has been sent to the unitysupport email address)

AndrewRH, 2021-09-02:
Thanks for your question.  Do you know what kind of HDR signal you are sending?  Currently we only support HLG and not HDR10 or others.

We hope to add for more HDR modes in the near future.  What are your requirements?

Cledwyn, 2021-09-23:
We're using HLG - Renderheads is set to Colourspace mode Rec2100. We have two cards providing input and output video. Rainbow/banding only appears when YCbCr 422 in and out is set to 10-bit (No artefacts using 8-bit - but that's not suitable for our use case).

Cledwyn, 2021-10-01:
Hi @AndrewRH - Is there an update on this item at all? Have you been able to reproduce the problem, using a combination of input and output cards?

AndrewRH, 2021-10-04:
Hi @Cledwyn 

Sorry we haven't had time to look into this yet due to other project commitments.

[image]

Can I ask whether you have the option 'Use HDR' selected, and whether choosing Rec 2020 instead of 2100 makes any difference?

Your intention is to generate HDR scenes in Unity and then output them in HLG format over SDI?

Thanks,

AndrewRH, 2021-11-02:
We're currently looking into this issue.

Thanks,

RenderHeadsMrT, 2021-11-05:
Hi @jessfdm-codes / @Cledwyn,

We've not been able to directly reproduce this issue but are trying to narrow down possibilities.

When HDR support was first implemented, it was on the basis that the hardware receiving the signal would be explicitly set to the matching colour space set in the plugin, i.e. if you set Rec 2020/2100 the frame data was encoded as such but still sent as a Rec 709 signal and it was down to the receiver to be forced to treat it appropriately.

Since then, support for sending HDR metadata alongside the signal has been provided and so we've just released v1.9.1 featuring this, so now, if the DeckLink hardware supports HDR metadata then if you set output to Rec 2020/2100 (HLG) in the plugin it will output metadata to indicate as such, avoiding potential incorrect conversions.

We'd love to know if this new version helps?

Cheers,

Chris-RH, 2022-05-13:
@RichRH 

Closed due to lack of response. Reopen if this is still an issue.

closed 2022-05-13 by Chris-RH