RenderHeads/UnityPlugin-AVProDeckLink #50: Plugin 1.9.0 not working any more with Windows 7

pcossinet, 2021-09-10:
**Describe the bug**
Using Plugin 1.8.6 Decklink output is OK on a Windows 7 config.
After 1.9.0 upgrade, no more SDI output visible, no log or error message but all frames are dropped.
Works really perfectly on Windows 10.

**Your Setup (please complete the following information):**
 - Unity version: 2020.3.12f1
 - AVPro DeckLink version: 1.9.0
 - Operating system version: Windows 7
 - Hardware model: HP Z240
 - Desktop Video version: 12.0
 - Video mode (resolution, frame-rate, pixel-format): 1980i50

**To Reproduce**
Steps to reproduce the behavior:
Just run on Windows 7

RenderHeadsMrT, 2021-11-05:
Hi @pcossinet,

Are you able to confirm if the just released v1.9.1 fixes this for you?

Cheers,

AndrewRH, 2022-03-23:
Closing, assumed fixed

closed 2022-03-23 by AndrewRH