RenderHeads/UnityPlugin-AVProDeckLink #51: Documentation not clear enough

leandrogonzaga, 2021-09-10:
Hello.

We are considering purchasing the full version.  However, we are having difficulties on configuring the components to accept multiple outputs. We used the documentation, but it was not clear enough to do so.

We have a DeckLink 4K Extreme, and needing to use 2 outputs for 2 virtual cameras on Unity.  However, when recording the two outputs, we are faced with errors.

 What would be the logical step-by-step for configuring multiple outputs?

 grateful in advance

AndrewRH, 2021-09-15:
What errors are you faced with?

Did you set the Device Profile option of the device to allow it to run in a configuration that support multiple outputs?

kahnivore, 2021-09-29:
Hi @leandrogonzaga 

Please let us know if you still need help with your setup.

closed 2021-10-06 by kahnivore