RenderHeads/UnityPlugin-AVProDeckLink #54: Is 4K 60fps hdr argb support?

oceancx, 2021-11-03:
We have tried 4k options to output our render result.
We want hdr option to be open so the camera's rendertexture format is R32G32B32A32_SFLOAT.
Besides, 4K,60fps and 8-bit ARGB pixel format is what we expect from this plugin.
But it doesn't work under these options.
It works two ways, one is 4K,30fps,8-bit ARGB , the other is 4K,60fps,8-bit UYVY 4:2:2. the first one is lack of fps, the second one makes the whole image too bright. none of them is perfect.
Is there a way to make what we want happen?

AndrewRH, 2021-11-04:
Yes it should all be supported.

Which card are you using?  It sounds like you're having bandwidth issues since 4K 60 RGBA using more bandwidth than the UYVY format.  It could also be your cable - we foudn that high bandwidth needs very good SDI cables.

Image is too bright?  That sounds odd - coudl you send a screenshot?  Which version of the plugin are you using?  Are you using Unity in Linear or Gamma ColorSpace (in Player Settings)?

Thanks,

oceancx, 2021-11-05:
1. we are using RTX3090 card, the SDI cable is the good one that support up to 6K output and we tried use OBS capture our final rendertexture( 3840x2160 ARGB_SFLOAT) to do SDI output and it worked, on the SDI in computer, the color format was identified YUV but no color difference.

2. the plugin version is (plugin v1.9.0 script v1.9.0) and color space is Linear. I notice maybe the color space is what caused the image is too bright, we will try to find out the result.

BTW, sounds that no matter which color format whether RGB or YUV, the final output's color should have no difference?

RenderHeadsMrT, 2021-11-05:
Hi @oceancx,

Thanks for the update - Which DeckLink card are you using? Are you just outputting or processing input here too? Also, are you running through the Unity editor or a build? (The editor can add significant overhead, especially if you're also viewing DeckLink previews in the inspector for example)

We'd expect YUV and RGB output to show no obvious difference (depending on bit-depth YUV can provide better compression at the expense of some colour precision) - if you could send any screenshots demonstrating issues after you've tried changing colour space it'd be much appreciated.

Cheers,

closed 2021-11-26 by stale[bot]