RenderHeads/UnityPlugin-AVProDeckLink #58: Enable 3d not working

PulsarCA, 2022-08-27:
**Describe the bug**
Video blanking when "Enable 3d" is checked.

I am outputting an HDMI video stream from Davinci Resolve via a Decklink Extreme 3d+ card, then re-capturing that stream in Unity with an UltraStudio 4k. I have tried SBS, Over/Under, and line by line formats. AVPro Decklink will recognize and display the video stream, and display them as mono SBS (etc) video but as soon as "Enable 3d" is checked the video blanks. The end goal is to be able to view a live stereoscopic video stream from Resolve in a VR environment.

**Your Setup (please complete the following information):**
 - Unity version: 2021.3.5f
 - AVPro DeckLink version: 1.9.3
 - Operating system version: Windows 10 Pro
 - Hardware model: Decklink Extreme 3d+ / UltraStudio 4k
 - Cables used: HDMI
 - Desktop Video version: 11.7
 - Video mode (resolution, frame-rate, pixel-format): 1080p, 23.98
 - DaVinci Resolve 17.4.6

**To Reproduce**
The problem exists in both my own project and the "Apply to Material" Demo Scene. The only error thrown is:

[AVProDeckLink] FormatConverter failed to create raw texture
UnityEngine.Debug:LogWarning (object)
RenderHeads.Media.AVProDeckLink.FormatConverter:Build () (at Assets/AVProDeckLink/Scripts/Internal/FormatConverter.cs:202)
RenderHeads.Media.AVProDeckLink.FormatConverter:Build (int,RenderHeads.Media.AVProDeckLink.DeviceMode,bool,bool) (at Assets/AVProDeckLink/Scripts/Internal/FormatConverter.cs:184)
RenderHeads.Media.AVProDeckLink.Device:StartInput (int,RenderHeads.Media.AVProDeckLink.AudioSettings,bool,bool) (at Assets/AVProDeckLink/Scripts/Internal/Device.cs:789)
RenderHeads.Media.AVProDeckLink.DeckLinkInput:BeginDevice () (at Assets/AVProDeckLink/Scripts/Components/DeckLinkInput.cs:209)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin (bool) (at Assets/AVProDeckLink/Scripts/Internal/DeckLink.cs:560)
RenderHeads.Media.AVProDeckLink.DeckLink:RenderExplorer () (at Assets/AVProDeckLink/Scripts/Internal/DeckLink.cs:273)
RenderHeads.Media.AVProDeckLink.DeviceExplorerManager:OnGUI () (at Assets/AVProDeckLink/Scripts/Internal/DeviceExplorerManager.cs:77)

However this error appears whether or not "Enable 3d" is checked and does not seem to affect performance.

Thanks for the help!

Chris-RH, 2022-08-31:
Hi @PulsarCA,

Our Decklink Dev is on holiday until the end of next week, so I'll get him to look into this as soon as he can.

RichRH, 2022-10-27:
Hi @PulsarCA,

Apologies it's taken a while but I've finally managed to look into this and found/fixed a couple of code issues that were preventing 3D from working on my test setup:

- DeckLink 4K Pro
- Unity 2021.3.11f1
- Build of 'Output Demo' with 'Enable 3D' checked and an additional camera added for the right eye
- Loopback of both card outputs into both card inputs
- 'Apply to Material' scene for input in editor with 'Enable 3D' checked
- Left eye now succesfully rendered to material(s)

All being well we aim to get a new build out next week with these fixes.

Cheers,

RichRH, 2022-11-08:
Hi @PulsarCA,

1.9.4 just released which hopefully fixes the above for you - do let us know if you're still seeing issues.

Cheers,

closed 2022-11-14 by Chris-RH