RenderHeads/UnityPlugin-AVProDeckLink #60: Choppy Animations on Decklink in 1080i50

303er, 2022-10-28:
Please DO NOT LINK / ATTACH YOUR PROJECT FILES HERE

**Describe the bug**
Decklink Output in 1080i50  is choppy and far from smooth. Animations in Game Window are smooth.

**Your Setup (please complete the following information):**
 - Unity version:  2021.3.10f1
 - AVPro DeckLink version: UnityPlugin-AVProDeckLink-v1.9.3-Trial
 - Operating system version: Windows 11
 - Hardware model: 
 - Cables used: HDMI/SDI
 - Desktop Video version: 12.4
 - Video mode (resolution, frame-rate, pixel-format): 1080i50 ARGB

**To Reproduce**
Steps to reproduce the behavior:
1.  Open External Keying Demo Scene
2. Insert Cube in Scene.
3. Make loop animation for cube moving horizontally on the screen.
4. Compile Project.
5. Playback will be choppy on Decklink output, even when its smooth in game window.

**Logs**
If applicable, add error logs to help explain your problem.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Videos**
If applicable, add a copy of your video or the URL

Please DO NOT LINK / ATTACH YOUR PROJECT FILES HERE
Instead email the link to us unitysupport@renderheads.com

RichRH, 2022-10-28:
Hi @303er,

Thanks for the clear reproduction steps - I have fixed a recent frame capture bug but will try and replicate your setup to see if I can track any further issues.

Cheers,

RichRH, 2022-11-03:
Hi @303er,

Just a quick update to say I believe I've located and fixed the issue - should be in the next build hopefully released by early next week.

Cheers,

closed 2022-11-08 by RichRH

303er, 2022-11-09:
> Hi @303er <https://github.com/303er>,
>
> 1.9.4 just released which hopefully fixes the above for you - will close
> this ticket but please do reopen if you're still seeing issues.
>
> Cheers,

Awesome.
It works great for me now.
We will purchase a license in the next coming days.

Cheers,
Roman