RenderHeads/UnityPlugin-AVProDeckLink #61: Timecode Incorrect above 30fps

BlueprintBen, 2022-11-28:
When reading out the timecode data from the camera, if the camera is set to output at 50fps or 60fps, the timecode's frame portion only goes up 24 or 29 (respectively) instead of 49/59. The timecode displayed on the camera feed itself correctly displays frame numbers up to 49/59. 

 - AVPro DeckLink version: 1.9.4
 - Operating system version: Windows 
 - Hardware model: Decklink 8K Pro + BlackMagic URSA 12K
 - Cables used: SDI
 - Desktop Video version: 12.1
 - Video mode: 3840x2160, 50/60fps, YUV 4:2:2
 
Steps to reproduce the behavior:
1. Modify an example scene to print out Timecodes from Device.GetFrameTimeCode() in HH:MM:SS.FF format
2. Set camera to output at 50 or 60 FPS.
3. Enable Timecode output from camera
4. Run the example and notice that the .FF portion of the timecode never goes above *half* of the camera's capture framerate.

Chris-RH, 2022-12-05:
We will look into this :)

RichRH, 2023-01-11:
Hi @BlueprintBen,

Thanks for reporting this - we have located and fixed this issue which will be in the next DeckLink release (hopefully a week or two's time).

Cheers,

Chris-RH, 2023-01-31:
This has been fixed in the current release, please let us know if this is still a problem for you.

closed 2023-02-07 by Chris-RH