RenderHeads/UnityPlugin-AVProDeckLink #63: 'ETC_RGB4' is not supported on this platform.

v2k, 2023-03-26:
Input fails, error spams

 - Unity version: 2022.2
 - AVPro DeckLink version: 1.95
 - Operating system version: Windows
 - DX11

```
 [AVProDeckLink] Changing device 'DeckLink 8K Pro (1)' input 1920x1080 (HD1080i 59.94 10-bit 4:2:2 YUV)Texture creation failed. 'ETC_RGB4' is not supported on this platform. Use 'SystemInfo.SupportsTextureFormat' C# API to check format support.
NullReferenceException
  at (wrapper managed-to-native) UnityEngine.Texture.set_wrapMode(UnityEngine.Texture,UnityEngine.TextureWrapMode)
  at RenderHeads.Media.AVProDeckLink.FormatConverter.CreateRawTexture () [0x0010c] in <e69ed9c0cab84e9891a0e0f3b09d885e>:0 
  at RenderHeads.Media.AVProDeckLink.FormatConverter.Build () [0x00008] in <e69ed9c0cab84e9891a0e0f3b09d885e>:0 
  at RenderHeads.Media.AVProDeckLink.FormatConverter.Build (System.Int32 deviceHandle, RenderHeads.Media.AVProDeckLink.DeviceMode mode, System.Boolean delayResourceCreationUntilFramesStart, System.Boolean useHdr) [0x00044] in <e69ed9c0cab84e9891a0e0f3b09d885e>:0 
  at RenderHeads.Media.AVProDeckLink.Device.ChangeInput (System.Int32 modeIndex, System.Boolean allowRetry) [0x000ae] in <e69ed9c0cab84e9891a0e0f3b09d885e>:0 
  at RenderHeads.Media.AVProDeckLink.Device.Update () [0x00097] in <e69ed9c0cab84e9891a0e0f3b09d885e>:0 
  at RenderHeads.Media.AVProDeckLink.DeckLink.Update () [0x0002e] in <e69ed9c0cab84e9891a0e0f3b09d885e>:0 
```

Chris-RH, 2023-03-29:
Can you explain what you did to get this? 
Has it stopped working since a recent update?

v2k, 2023-03-29:
I believe this happens with DX11.  In FormatConverter.cs I had to comment out the "textureFormat = TextureFormat.ETC_RGB4" line below.

```
#if UNITY_2022_1_OR_NEWER
					// RJT NOTE: It seems the format parameter is mostly ignored, except under Unity 2022(+?)/D3D12, where it appears to create
					// a 'default' SRV for it, but unfortunately there isn't an obvious 'TextureFormat' for our native 'DXGI_FORMAT_R10G10B10A2_UNORM'
					// format and it crashes, so for now forcing it to a format that appears to bypass invalid SRV creation..
					// textureFormat = TextureFormat.ETC_RGB4;//DXGI_FORMAT_R10G10B10A2_UNORM;
#endif
```

And

```
				// Account for potential padding
				TextureFormat textureFormat = TextureFormat.ARGB32;
				if (_mode.PixelFormat == DeckLinkPlugin.PixelFormat.YCbCr_10bpp_422)
				{
					textureWidth = ((_mode.Width + 47) / 48) * 32;
#if UNITY_2022_1_OR_NEWER
					// RJT NOTE: It seems the format parameter is mostly ignored, except under Unity 2022(+?)/D3D12, where it appears to create
					// a 'default' SRV for it, but unfortunately there isn't an obvious 'TextureFormat' for our native 'DXGI_FORMAT_R10G10B10A2_UNORM'
					// format and it crashes, so for now forcing it to a format that appears to bypass invalid SRV creation..
					// textureFormat = TextureFormat.ETC_RGB4;//DXGI_FORMAT_R10G10B10A2_UNORM;
#endif
				}
				else if (_mode.PixelFormat == DeckLinkPlugin.PixelFormat.RGB_10bpp_444)
				{
					textureWidth = ((_mode.Width + 63) / 64) * 64;
#if UNITY_2022_1_OR_NEWER
					// textureFormat = TextureFormat.ETC_RGB4;//DXGI_FORMAT_R8G8B8A8_UINT;
#endif
				}

```

v2k, 2023-03-29:
This was from a clean install of 1.9.5 into our Unity2022.2 project.

RichRH, 2023-03-30:
Code changed as per https://github.com/RenderHeads/UnityPlugin-AVProVideo/issues/1286 which seems to remove errors and permit builds.

Cheers,

closed 2023-03-30 by RichRH