RenderHeads/UnityPlugin-AVProDeckLink #64: NullReferenceException: Object reference not set to an instance of an object

v2k, 2023-03-26:
Spams this error in log.  

 - Unity version: 2022.2
 - AVPro DeckLink version: 1.9.5
 - Operating system version: Windows 10
 - DX11
 - SDI input 2 subdevice full duplex

```
NullReferenceException: Object reference not set to an instance of an object
  at RenderHeads.Media.AVProDeckLink.DeckLinkOutput.CaptureFrame () [0x000fd] in C:\Users\HDR-Tester\Documents\dev\trunk\UnityProject\Assets\AVProDeckLink\Scripts\Components\DeckLinkOutput.cs:1038 
  at RenderHeads.Media.AVProDeckLink.DeckLinkOutput.Process () [0x003df] in C:\Users\HDR-Tester\Documents\dev\trunk\UnityProject\Assets\AVProDeckLink\Scripts\Components\DeckLinkOutput.cs:1341 
  at RenderHeads.Media.AVProDeckLink.DeckLink.Update () [0x0002c] in C:\Users\HDR-Tester\Documents\dev\trunk\UnityProject\Assets\AVProDeckLink\Scripts\Internal\DeckLink.cs:590 

```

Chris-RH, 2023-03-29:
Can you explain what you did to get this? 
Has it stopped working since a recent update?

v2k, 2023-03-29:
Clean install of 1.9.5 on Unity2022.2 project.

Had to add a null check below in DeckLinkOutput.cs:

```
		else
			{
				for (int i = 0; i < 2; i ++)
				{
					PerEye eye = _eye[i];
					if (IsEyeUsed(i) && eye != null && eye.blended != null)
					{
						eye.blended.DiscardContents();
						Graphics.Blit(eye.inputTexture, eye.blended);
					}
				}
			}
		}
```

vs

`if (IsEyeUsed(i))`

RichRH, 2023-03-31:
Hi @v2k,

This one's a bit of a mystery as a) '_eye' is allocated as part of the 'DeckLinkOutput' class ('private PerEye[] _eye = new PerEye[2];') and b) '.blended' is allocated as part of the 'DeckLinkOutput::BeginDevice()' -> 'InitCaptureBlendResources()' call which also sets up the variable ('IsStreamingOutput') that allows the above code you've highlighted to actually be called.

Is this reproducible in one of our demos? (I've tried to no avail thus far) and if not, are you maybe able to add some breakpoints  (on the sections I mention above) to confirm how things could be running out of order for you?

Cheers,

Chris-RH, 2023-05-08:
any more info?

v2k, 2023-05-08:
We've been running with this hack.  I tried removing the hack and the issue seems to not be happening now.  That said, I'm testing on 1 sub full duplex at the moment because of the other issue related to 2 sub full duplex.

Chris-RH, 2023-05-09:
ok thank you, that's interesting to know

closed 2023-05-15 by v2k