RenderHeads/UnityPlugin-AVProDeckLink #65: Two Sub Device Full duplex

v2k, 2023-03-28:
How does one setup Two Sub Device Full duplex?  It looks like multi output has been removed as an option from 1.9.5

Trying to setup with an input device and 2 output devices as below.  

input 1: SDI 1: DeckLink Pro 8K (1)

output 1 keying none: SDI 2: DeckLink Pro 8K (2) 
output 1 keying external: SDI 3/4: DeckLink Pro 8K (3)

GameObjects: all set to 
DeckLinkInput - on index 0
DeckLinkOutput - on index 1 
DeckLinkOutput - on index 2 - does not find device, nor does it show up in explorer

 - Unity version: 2022.2
 - AVPro DeckLink version: 1.9.5
 - Operating system version: Win10
 - Cables used: in SDI 1, out on SDI 2, and 3/4
 - Desktop Video version: 12.4

DeckLink 8K Pro (3) and DeckLink 8K Pro (4) do not show up in the explorer:
[image]

[image]

v2k, 2023-03-28:
Sorry, 2 devices is correct, but what I want in the scene is something like:

Device 1: DeckLink 8K Pro (1) in/out (SDI 1/SDI 2)
Device 2: DeckLink 8K Pro (2) out (SDI 3 + 4)

decklink input external on DeckLink 8K Pro (1)
decklink output external on DeckLink 8K Pro (1)

SDI 1: decklink input (Device 1)
SDI 2: decklink output (Device 1)

decklink output external on DeckLink 8K Pro (2)
SDI3: key (Device 2)
SDI4: fill (Device 2)

Not sure how to set this up with AVPro DeckLink.  It's also confusing that each DeckLinkInput and DeckLinkOutput component has its own Device Profile setting.  That's a global setting on the card itself, why is this on each component?  Seems like this should be all configured via the DeckLinkSettings component?

Chris-RH, 2023-03-30:
hmm, we've done it before, just trying to remember how :D

RichRH, 2023-03-31:
Hi @v2k,

Before I look further into this, yeah, I agree that the device profile settings should be global within the plugin - I believe this may be legacy catching up with us. What I tend to do is use Blackmagic's 'Desktop Video Setup' app to setup the profile I require then just use device indices on the inputs/outputs within our plugin.

Cheers,

v2k, 2023-03-31:
I would rather have the plugin set the profile on the device for the application; especially if each application has its own device profile.  i.e. set the Blackmagic card's profile, then configure each subdevice.  Ideally one would use Desktop Video Setup to verify and debug; not to configure the card's profile.

It would be nice if the "Device Selection" on the input and output components mapped to the subdevice names; or were pulled from Blackmagic instead of searching by name/index.

v2k, 2023-05-04:
Do you have a method for this?  I have it running in 2 SubDevice Full Duplex, and it works with keying None.  But if I set one of the outputs to External keying, it seems to break.

2 sub full duplex; 2 keying None outputs:
[image]

2 sub full duplex; 1 keying None output, 1 keying External output:
[image]

v2k, 2023-05-05:
I was able to repro this in your sample; indeed 2 Sub Full Duplex does not work with External key.  It works with 1 Sub Full Duplex.  Here are the slightly modified sample scenes to show this:

[SampleExternal2SubNoGood.zip](https://github.com/RenderHeads/UnityPlugin-AVProDeckLink/files/11408842/SampleExternal2SubNoGood.zip)

v2k, 2023-05-10:
Curious if anyone is able to investigate this issue?

RichRH, 2023-07-03:
Hi @v2k,

Apologies it's taken a while to get back to you.

Thanks for providing your modified sample scene - I downloaded and tested "TwoSubFull" on our DeckLink 8K Pro and in its current form I only get input into SDI 1, internally looped-through to output on SDI 2 (default behaviour as expected) and nothing on SDI 3 or 4.

However, I spotted that the 'Device Index' for the 'Output' device was set to 0 (the same as the 'Input' device) and changing that to 1 provided the expected output - key on SDI 3 and fill on SDI 4.

Does this work for you?

Cheers,

v2k, 2023-07-08:
Hey @RichRH 

Thanks for looking into this.  It looks like that may have been related to the issue.  External keying needs two SDI ports, and if Device 0 is input and output, there is only 1 for output.  So it can only do None for keying.  I added a second output for Device 1 with External keying and it appears promising.  I'll do further testing with this.

[image]
[image]

v2k, 2023-07-08:
[image]
Here's the setup I have now:

Input: Device 0
Output: Device 0 (Keying None)
OutputKeyFill: Device 1 (Keying External)

I ran into some spamming errors with the multiple outputs.  In DeckLinkManager.cs I had to comment out Ln 327 to false:

```
		void LateUpdate()
		{
#if true
			// If multiple outputs are active then attempt to sync Unity to them
			// RJT TODO: Would be far nicer to have a single point of control for Unity update speed so could
			// look at migrating non-multi-output/'DeckLinkOutput::SetUpdateFrameRate()' here too?
			if (DeckLinkSettings.Instance._multiOutput)
			{
	#if false // true	// Lowest Common Multiple (LCM) of all output rates
				{
					// Determine LCM (if multiple outputs)
					// RJT NOTE: We also add (/subtract) any changes made in 'DeckLinkOutput::AdjustPlaybackFramerate()'
					// to compensate for buffers becoming too empty (/full) but don't factor this into the LCM or
					// target framerate could be all over the place!
					int targetFramerate = -1, compensateFramerate = 0;
					foreach (DeckLinkOutput output in _outputs)
					{
						if ((output.Device != null) && output.Device.IsStreamingOutput)
						{
							targetFramerate = ((targetFramerate < 0) ?
								output./*OutputFramerate*/TargetFramerate :
								SCD(targetFramerate, output./*OutputFramerate*/TargetFramerate)
							);

							compensateFramerate += (output.OutputFramerate - output.TargetFramerate);
						}
					}

					// Lock Unity to this rate
					Application.targetFrameRate = (targetFramerate + compensateFramerate);
				}
	#elif false	// ?
				{

				}
	#endif
			}
#endif
		}

```

v2k, 2023-07-10:
Upon more testing, this works, but decimates performance.  I'm testing on a 1080i@5994 signal.  It's pretty much unusable with multi-outputs.  I tried with and without setting SyncToInput on the outputs.

Disabling or removing either of the 2 output components restores performance.  

With the sample, it might be too lightweight to notice the performance impact.   The health status goes from solid green (with a single output) to flickering from green to yellow with the configuration described above.

v2k, 2023-07-10:
Found a bug causing a restart of an output device, this seems to have resolved it.

closed 2023-07-10 by v2k