RenderHeads/UnityPlugin-AVProDeckLink #66: How to : HDR ?

pcossinet, 2023-07-10:
Today we work with AvPro decklink in 1080i50, 1080p50 & 20160p50 with Decklink duo 2 & Decklink 8K and all is ok, the plugin is awesome.

We are looking for the future: HDR

Our Dev Setup
- Unity 2020.3.12f1
- AvPro Decklink 1.9.5
- Decklink API 12.2.0.0
- Decklink UltraStudio 4K Mini
- Decklink Video Assist 7" HDR 12G

With a clean Project we added a Camera and Avpro Decklink Output
common setting:
- HD1080p
- FrameRate 50
- No Interlaced
- Use Hdr enabled

Test 1: 8-bit UYVY 4:2:2 / Rec 709
> Output ok
> No HDR icon on Monitor
> Rec.709 indicator

[image]

Test 2: 8-bit UYVY 4:2:2 / Rec 2020
> Output ok
> No HDR icon on Monitor
> Rec.709 indicator

[image]

Test 3: 8-bit UYVY 4:2:2 / Rec 2100
> Output ok
> HDR icon on Monitor
> Rec.709 - HLG indicator

[image]

Test 4: 10-bit UYVY 4:2:2 / Rec 2100
> Output Dark
> HDR icon on Monitor
> Rec.709 - HLG indicator

8-bit RGBA & 8-bit ARGB works like 8-bit UYVY
10-bit RGB, RGBX & RGBX LE doesn't output anything

So we are a little confuse about the right utilisation of the plugin for HDR output.

What we expected is:
- 10-bit format
- enabled HDR
- select the rightColor Space with the BroadCast partner

But the only good setting with this setup is:
Setup 3: 8-bit UYVY 4:2:2 or 8-bit RGBA / Rec 2100
> Output ok
> HDR icon on Monitor
> Rec.709 - HLG indicator

What is in your opinion the good settings or requested version or setup to have a true HDR output ?

Thanks for your feedback

RichRH, 2023-07-17:
Hi @pcossinet,

Thanks for your feedback and the detailed report - I'm looking into this to see if I can spot any other oddities but observations/feedback so far:

- You state 'No Interlaced' but the monitor is reporting '1080i50' in your grabs - I've tested here and toggling the interlaced flag seems to correctly report 1080i50/1080p50 (just wondering if you're seeing a bug)
- I'm testing with a new scene like yours, Decklink 8K Pro into Video Assist 7" HDR 12G over SDI (1080p50)
- 'Use Hdr' flag is required for capture textures to be treated as floating point (and thus permit HDR ranges)
- Rec. 709, 2020 and 2100 appear to output and be correctly detected by the Video Assist with YUV formats (NOTE: 2100 is reported as '2020 - HLG')
- Only Rec. 2100 supports HDR output and we currently only support the HLG transfer function - 8 and 10-bit YUV seem to output and be detected as expected, although 8-bit shouldn't really be used with HDR
- I've located a bug with 10-bit RGB formats which indeed prevents output - will look into fixing ASAP, so currently Rec. 2100 10-bit YUV is your best bet for HDR
- We have some improvements on our TODO list including HDR10/PQ support and 12-bit formats

Cheers,

RichRH, 2023-07-31:
With the release of 1.9.6 (fix for 10-bit RGB not outputting) I'm changing this ticket to an 'enhancement' for the 'HDR10/PQ support and 12-bit formats' TODOs