RenderHeads/UnityPlugin-AVProDeckLink #69: Crash when starting multiple inputs

pilzinho, 2023-10-23:
We get crashes when starting the application with multiple active inputs. Up to 2 inputs it seems to work fine but starting from 3 inputs we get regular crashes. Tested on multiple Windows machines with different Full HD/4K inputs. So it seems to get worse when using more inputs and higher resolutions. We also reproduced it with the InputExplorerDemo. We get around 1 crash when starting/stopping the application 10 times.

* Unity version: 2022.3.3f1
* AVPro DeckLink version: 1.9.6
* Operating system version: Windows 10
* Hardware model: DeckLink Quad HDMI Recorder
* Desktop Video version: 12.5.1
* Input example:
  * 4096x2160 (4K DCI 60 10-bit 4:2:2 YUV)
  * 1920x1200 (PC 1920x1200 60 10-bit 4:4:4 RGB)
  * 3840x2160 (4K 29.97 10-bit 4:4:4 RGB)
  * 3840x2160 (4K 60 10-bit 4:2:2 YUV)

We always have one of the 2 following stack traces:

```
(ntdll) RtlWaitOnAddress
(ntdll) RtlEnterCriticalSection
(ntdll) RtlEnterCriticalSection
(AVProDeckLink) GetHealthStatus
(AVProDeckLink) GetHealthStatus
(AVProDeckLink) GetHealthStatus
(Mono JIT Code) (wrapper managed-to-native) RenderHeads.Media.AVProDeckLink.DeckLinkPlugin:GetTexturePointer (int)
(Mono JIT Code) RenderHeads.Media.AVProDeckLink.FormatConverter:CreateRawTexture ()
(Mono JIT Code) RenderHeads.Media.AVProDeckLink.FormatConverter:Build ()
(Mono JIT Code) RenderHeads.Media.AVProDeckLink.FormatConverter:Build (int,RenderHeads.Media.AVProDeckLink.DeviceMode,bool,bool)
(Mono JIT Code) RenderHeads.Media.AVProDeckLink.Device:ChangeInput (int,bool)
(Mono JIT Code) RenderHeads.Media.AVProDeckLink.Device:Update ()
(Mono JIT Code) RenderHeads.Media.AVProDeckLink.DeckLink:Update ()
```

or

```
0x00007FF9E2CB34F6 (ntdll) RtlWaitOnAddress
0x00007FF9E2C7FCB4 (ntdll) RtlEnterCriticalSection
0x00007FF9E2C7FAE2 (ntdll) RtlEnterCriticalSection
0x000000018000F40A (AVProDeckLink) GetHealthStatus
0x000000018000F8F8 (AVProDeckLink) GetHealthStatus
0x000000018000F382 (AVProDeckLink) GetHealthStatus
0x0000028D46D5AA4A (Mono JIT Code) (wrapper managed-to-native) RenderHeads.Media.AVProDeckLink.DeckLinkPlugin:GetTexturePointer (int)
0x0000028D46D695B3 (Mono JIT Code) RenderHeads.Media.AVProDeckLink.FormatConverter:Update ()
0x0000028D46D690D3 (Mono JIT Code) RenderHeads.Media.AVProDeckLink.Device:Update ()
0x0000028D46D67A23 (Mono JIT Code) RenderHeads.Media.AVProDeckLink.DeckLink:Update ()
```

RichRH, 2023-10-30:
Hi @pilzinho,

Just to confirm, does this happen in editor when pressing the 'play' button or is there more to it?

Cheers,

pilzinho, 2023-11-02:
Hi! Thanks for the reply. We tested with Windows x64 builds of our application and the InputExplorerDemo. But it happened in the editor as well as with the builds.

BlueprintBen, 2023-11-06:
I've got the same call stack as your second - coming from intermittent crashes upon starting even just 1 input, provided that it was previously started and stopped. Similarly would say that the crash occurs about 1/10 input starts.

```
Stacktrace:

  at <unknown> <0xffffffff>
  at (wrapper managed-to-native) RenderHeads.Media.AVProDeckLink.DeckLinkPlugin.GetTexturePointer (int) [0x00009] in <ad493bb82a764e9eaa8fc85ddedef005>:0
  at RenderHeads.Media.AVProDeckLink.FormatConverter.Update () [0x0000d] in C:\Projects\mixcast-service\Assets\Addons\AVProDeckLink\Scripts\Internal\FormatConverter.cs:220
  at RenderHeads.Media.AVProDeckLink.Device.Update () [0x0011c] in C:\Projects\mixcast-service\Assets\Addons\AVProDeckLink\Scripts\Internal\Device.cs:1065
  at RenderHeads.Media.AVProDeckLink.DeckLink.UpdateManual () [0x00042] in C:\Projects\mixcast-service\Assets\Addons\AVProDeckLink\Scripts\Internal\DeckLink.cs:597
  at RenderHeads.Media.AVProDeckLink.DeckLinkInput.Update () [0x00001] in C:\Projects\mixcast-service\Assets\Addons\AVProDeckLink\Scripts\Components\DeckLinkInput.cs:123
  at (wrapper runtime-invoke) object.runtime_invoke_void__this__ (object,intptr,intptr,intptr) [0x00020] in <695d1cc93cca45069c528c15c9fdd749>:0

=================================================================
Got a SIGSEGV while executing native code. This usually indicates
a fatal error in the mono runtime or one of the native libraries 
used by your application.
=================================================================

Received signal SIGSEGV
Stack trace:
0x00007ffa6ef918d4 (ntdll) RtlEnterCriticalSection
0x00007ffa6ef916c2 (ntdll) RtlEnterCriticalSection
0x0000021d020df37a (AVProDeckLink) GetHealthStatus
0x0000021d020df908 (AVProDeckLink) GetHealthStatus
0x0000021d020df2f2 (AVProDeckLink) GetHealthStatus
0x0000021d903e3f3a (Mono JIT Code) (wrapper managed-to-native) RenderHeads.Media.AVProDeckLink.DeckLinkPlugin:GetTexturePointer (int)
0x0000021d904baa33 (Mono JIT Code) [FormatConverter.cs:220] RenderHeads.Media.AVProDeckLink.FormatConverter:Update () 
0x0000021d904b9d23 (Mono JIT Code) [Device.cs:1065] RenderHeads.Media.AVProDeckLink.Device:Update () 
0x0000021d904b599b (Mono JIT Code) [DeckLink.cs:597] RenderHeads.Media.AVProDeckLink.DeckLink:UpdateManual () 
0x0000021d904b5653 (Mono JIT Code) [DeckLinkInput.cs:123] RenderHeads.Media.AVProDeckLink.DeckLinkInput:Update () 
0x0000021cb5361ef8 (Mono JIT Code) (wrapper runtime-invoke) object:runtime_invoke_void__this__ (object,intptr,intptr,intptr)
0x00007ff9d8b600e0 (mono-2.0-bdwgc) [mini-runtime.c:2849] mono_jit_runtime_invoke 
0x00007ff9d8ae2ac2 (mono-2.0-bdwgc) [object.c:2921] do_runtime_invoke 
0x00007ff9d8aebb1f (mono-2.0-bdwgc) [object.c:2968] mono_runtime_invoke 
0x00007ff64e3c4a94 (Unity) scripting_method_invoke
0x00007ff64e3bfb91 (Unity) ScriptingInvocation::Invoke
0x00007ff64e38d344 (Unity) MonoBehaviour::CallMethodIfAvailable
0x00007ff64e38d432 (Unity) MonoBehaviour::CallUpdateMethod
0x00007ff64deea498 (Unity) BaseBehaviourManager::CommonUpdate<BehaviourManager>
0x00007ff64def06fa (Unity) BehaviourManager::Update
0x00007ff64e0ee57d (Unity) `InitPlayerLoopCallbacks'::`2'::UpdateScriptRunBehaviourUpdateRegistrator::Forward
0x00007ff64e0d557a (Unity) ExecutePlayerLoop
0x00007ff64e0d56e9 (Unity) ExecutePlayerLoop
0x00007ff64e0dc6d9 (Unity) PlayerLoop
0x00007ff64edb5841 (Unity) PlayerLoopController::UpdateScene
0x00007ff64edb3a6c (Unity) Application::TickTimer
0x00007ff64f226491 (Unity) MainMessageLoop
0x00007ff64f22a631 (Unity) WinMain
0x00007ff6508f71ae (Unity) __scrt_common_main_seh
0x00007ffa6e3e257d (KERNEL32) BaseThreadInitThunk
0x00007ffa6efcaa78 (ntdll) RtlUserThreadStart
```

* Note that the function DeckLink.UpdateManual () is just a slight tweak to the code to allow for more selective updating of the component.

RichRH, 2023-12-04:
Thanks for the input all - I'm merging this ticket into #68 as they're the same fundamental issue. A fix is currently in testing.

Cheers,

closed 2023-12-04 by RichRH

Chris-RH, 2023-12-21:
This issue has been fixed in the latest release (AVPro Decklink version 1.9.7). Please let us know if this issue has not been fixed for you.