RenderHeads/UnityPlugin-AVProDeckLink #70: 2 capture cards: Inputs change after restart

pilzinho, 2023-11-06:
We have an installation at a client using 2 capture cards. Sometimes after restarting the computer the inputs are switched so that the input indices start at the other card. This is a pretty severe problem as our application cannot just be reconfigured after a restart and the application won't work correctly anymore if the inputs are switched. We would need support for this issue as soon as possible. So any help would be greatly appreciated if you know of any way how to make the inputs work consistently using multiple capture cards.

Unity version: 2022.3.3f1
AVPro DeckLink version: 1.9.6
Operating system version: Windows 10
Hardware model: 2x DeckLink Quad HDMI Recorder
Desktop Video version: 12.5.1

RichRH, 2023-11-21:
Hi @pilzinho,

Instead of using indices, you might be able to use custom names - in Desktop Video Setup if you select each device, go to 'About' and change the 'Label' this should become the new device name:

[image]

Then, on the 'DeckLinkInput' in Unity for each device set the corresponding 'Device Name', and make sure the 'Device Index' is set to 0 as this index is into the now name filtered list:

[image]

Hopefully this helps?

Cheers,

pilzinho, 2023-11-30:
Hi, thanks for the tip! I currently only have one card for testing so I don't know yet if this resolves our issue with 2 cards getting mixed up. It will probably be another week until we know. But I was able to address the inputs of one card by custom names as you described. We're doing everything in code, so this is what I did. Would this be the correct way to do it?
```
GameObject decklinkObject = new GameObject();
DeckLinkInput input = decklinkObject.AddComponent<DeckLinkInput>();
input._exactDeviceName = true;
input._desiredDeviceName = "CustomDevice 0";
input._filterDeviceByIndex = false;
input._filterDeviceByName = true;
input.DeviceIndex = 0;
input.ModeIndex = 0;
input._playOnStart = true;
input._autoDeinterlace = true;
input._autoDetectMode = _autoDetect;
input.Begin(true);
```

RichRH, 2023-12-04:
Hi @pilzinho,

No worries - yes, that code looks to be correct (you could also set '_filterDeviceByIndex' to 'true' and it would in theory pick up the same device as long as '_desiredDeviceIndex' was also '0' - basically the first device based on your pre-filtered name match).

Closing this ticket for now but please do re-open if it doesn't work as expected.

Cheers,