RenderHeads/UnityPlugin-AVProDeckLink #71: Sample Of The Internal Keying Doesn't Work

YarikNikCode, 2023-11-22:
**Describe the bug**
The sample of the internal keying doesn't work. We've tried to customized it and set different settings, but still have got the same. Then, we've tried the C++ official example of the keying from Blackmagic on the same setup and it works fine.

**Our Setup (please complete the following information):**
 - Unity version: 2022.3.2f1
 - AVPro DeckLink version: 1.9.6
 - Operating system version: Windows 11 22H2
 - Hardware model: Decklink 8K Pro
 - Cables used: -
 - Desktop Video version: 12.5.1
 - Video mode (resolution, frame-rate, pixel-format): 4K DCI 60, 8 bit, 4:2:2 YUV

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Folder 'Demos'
2. Go To Scene '06_InternalKeyingDemo'
3. Click on 'Play' Butotn
5. See error (plugin cut the background, but not insert the input image)

RichRH, 2023-12-05:
Hi,

Thanks for posting this - have you made sure you've forced the 'DeckLinkOutput' component resolution/format to _exactly_ that of your incoming feed? Keying is finicky and won't work if they don't fully match (you'll just get the Unity main camera background instead - dark grey).

I tested here (albeit at 1080) and got the same results as you, unless the forced resolution/format in 'DeckLinkOutput' matched the input - I think the official C++ example forces the output signal to that of the input signal hence why it works.

[image]

Cheers,

YarikNikCode, 2023-12-05:
Hi Richard,

thank you vey much for the information, it helped and now everything works properly!

Cheers,

RichRH, 2023-12-05:
That's great @YarikNikCode - thanks for confirming that worked.. :)

Cheers,

closed 2023-12-05 by RichRH