RenderHeads/UnityPlugin-AVProDeckLink #73: Multiple Outputs Stops VFX and Physics.

Caramel-Andrew, 2024-01-09:
Sorry if this is not a bug, but I have not yet been able to get this to work even with the demo.

I have two inputs and two outputs. All are working fine. However, when I play any VFX they fail to animate.
Actual animations work fine, but Shuriken systems stick on the first frame.
If I delete or switch off one of the two outputs, the VFX work fine.

As a test I also added a cube with a RigidBody to the scene. With one output it correctly falls with gravity. With two outputs, it hangs in its start position.

**Your Setup (please complete the following information):**
 - Unity version: 2023.2.3f1
 - AVPro DeckLink version: 1.9.7
 - Operating system version: Windows 11
 - Hardware model: Blackmagic Design Decklink Quad 2
 - Cables used: SDI
 - Desktop Video version: No idea.
 - Video mode (resolution, frame-rate, pixel-format):  HD1080p 29.97 8-Bit-UYVY 4:2:2 (but some results with any setting)
- Additional: Unity URP project.
- Note: Docs say to set the multi cam option in the settings. However, this is no longer in the settings. Looking at the code, this seems to be automatic now.

**To Reproduce**
Steps to reproduce the behaviour:
Create a new project in Unity.
Add AVPro DeckLink asset.
Open Demo Output Scene.

Duplicate the output components (DeckLink and Camera).
Slightly move the second camera so it can be seen easily.

Stick a shuriken particle system in the scene in view of both cameras.
Add a cube with a rigid body somewhere so it can fall.

Run the project.
For me, the project runs and the animation of the robo arm works as expected. Frame rate counter goes up as expected. (Though materials have to be updated to URP). 
However, partcicle/shuriken system does not animate and the cube does not fall.

*Again, sorry if this is not a bug and I just set something up wrong.*

RichRH, 2024-01-09:
Hi @Caramel-Andrew,

Thanks for reporting this - I've had a quick look and it appears to be part-bug, part-change in Unity behaviour. In 2021 the above works fine but indeed, in 2023 it behaves as you state (unsure about 2022 as haven't had chance to check yet).

A workaround for the moment is to make sure DeckLink script calls to 'Time.captureFramerate' (generally setting to -1.0f) are commented out - hope that helps?

Cheers,

Caramel-Andrew, 2024-01-09:
Hi @RichRH ,

Thank you so much. The workaround you suggested has fixed the problem!

RichRH, 2024-01-10:
Hi @Caramel-Andrew,

Great, thanks for letting us know - I'll make sure this fix is in the next release (unsure when that will be yet though).

Cheers,

RichRH, 2024-12-19:
Closing this ticket with the release of https://github.com/RenderHeads/UnityPlugin-AVProDeckLink/releases/tag/1.9.8.

Cheers,

closed 2024-12-19 by RichRH