RenderHeads/UnityPlugin-AVProDeckLink #75: Audio input is gone if input is stopped once

pilzinho, 2024-01-15:
When stopping an input calling DeckLinkInput.StopInput(), the AudioClip of the AudioSource is destroyed. So when starting the input again by calling Begin() the audio will not start again.

We solved this for now in DeckLinkInput by destroying the whole AudioSource when stopping and moving the creation of the AudioSource from the Awake() to the Begin() method like so:

```
protected override void BeginDevice()
{
        ...
        
	if (_device != null)
	{
		_enable3D = _device.Enable3DInput;
		_device.FlipInputX = _flipX;
		_device.FlipInputY = _flipY;
		InitAudio();
		DeckLinkPlugin.FlushAudioBuffer(_deviceIndex);
	}
}

private void InitAudio()
{
	if (_audioSource)
		StopAudio();
	
	_audioSource = gameObject.AddComponent<AudioSource>();
	_audioSource.hideFlags = HideFlags.HideInInspector;
	_audioSource.loop = true;
}
		 
private void StopAudio()
{
	if (!_audioSource)
		return;
	
	_audioSource.Stop();
	if (_audioSource.clip != null)
	{
		AudioClip.Destroy(_audioSource.clip);
		_audioSource.clip = null;
	}
	Destroy(_audioSource);
}
```

StopAudio() is called by the StopInput() and Cleanup() methods.

RichRH, 2024-10-03:
Hi @pilzinho,

Thanks for reporting this (some time ago!) - I've implemented these changes for the next release.

Cheers,

RichRH, 2024-12-19:
Closing this ticket with the release of https://github.com/RenderHeads/UnityPlugin-AVProDeckLink/releases/tag/1.9.8.

Cheers,

closed 2024-12-19 by RichRH