RenderHeads/UnityPlugin-AVProDeckLink #81: [Bug] v1.9.7 Soak testing failing with output frame drops

v2k, 2024-12-13:
### Unity Version

2022.3.43f1

### AVPro DeckLink edition

Full

### AVPro Decklink Version

1.9.7

### Desktop Video version

14.3

### Which Windows version(s) are you using?

Windows 10

### Which graphics API(s) are you using?

Direct3D 11

### Decklink device model

DeckLink 8k Pro

### Cables used

SDI

### Video mode

Windowed

### The issue

Getting frame drops on output.  Counter after roughly 4 hours of running:

[image]

Match output mode to input was false in this test

v2k, 2024-12-13:
[Player 4.log](https://github.com/user-attachments/files/18132491/Player.4.log)

RichRH, 2024-12-19:
Hi @v2k,

We've just released https://github.com/RenderHeads/UnityPlugin-AVProDeckLink/releases/tag/1.9.8 which features some improvements to output frame drops, especially with full-duplex - hopefully this helps!

Cheers,

closed 2025-08-04 by Chris-RH