RenderHeads/UnityPlugin-AVProDeckLink #82: [Bug] output devices failing to enumerate still not working in 1.9.9

richard12121212, 2025-01-22:
### Unity Version

2022.3.54

### AVPro DeckLink edition

Trial

### AVPro Decklink Version

1.9.9

### Desktop Video version

14.3.0.0

### Which Windows version(s) are you using?

Windows 11

### Which graphics API(s) are you using?

Direct3D 11

### Decklink device model

Blackmagic Design DeckLink Quad HDMI Recorder
PCIe videokaart

### Cables used

HBMI

### Video mode

Different ones. but it does not start cause of the enumration f the videos.

### The issue

In the latestedversion we still have the outputs not being enumerated. while it should be fixed according the realease notes. 

 Version 1.9.9 - 9 January 2025 
 ○  Fixes 
 ■  Fixed issue with output devices failing to enumerate with certain 
 profiles 

### Log output

```shell

```

RichRH, 2025-01-23:
Hi @richard12121212,

Thanks for reporting this - strange as this definitely fixed the issue for the person who found it (outside GitHub). Out of interest, and if possible, have you tried an SDI card/connection and/or another machine?

We have that same card in another PC so I'll test when I can get access.

Cheers,

richard12121212, 2025-01-23:
Sorry then i read the fix to fast.. It does not get the inputs, which it did in earlier versions.

richard12121212, 2025-01-23:
int numDevices = DeckLink.GetNumDevices();
            log("[CameraRecordMenuButtonController]showDevicesINUM DeckLink TOTALDEVICES  numDevices=" + numDevices);

          returns now 0.

RichRH, 2025-01-23:
Ok, that's strange as the input code hasn't changed recently. I've just tried feeding an extra Windows (11) display into our Quad HDMI Recorder and our input demo at least seems to work fine:

[image]

Have you tried the input demo?

Cheers,

richard12121212, 2025-01-23:
We did not try the input demo. When run it for half year now. so we did roll back to a version from a month ago then it works.

RichRH, 2025-01-24:
Just to confirm you're saying 1.9.8 _does_ work then?

I've just tried breakpointing our 04_InputExplorerDemo which uses the same code as you mention above 'int numDevices = DeckLink.GetNumDevices();' and I'm seeing 4 devices as expected with latest code:

[image]

Would you be able to test to see if you're seeing different?

richard12121212, 2025-01-24:
we used one version older, see the logs below. Maybe was change between version 1.9.7 and 1.9.9

NOT WORKING

[AVProMovieCapture] Init version: 5.3.3 (plugin v5.3.3f1-ultra) with GPU NVIDIA RTX A4000 Direct3D 11.0 [level 11.1] OS: Windows 11  (10.0.22631) 64bit
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProMovieCapture.CaptureBase:Awake()
UnityEngine.Object:Instantiate(T)
 
 

[AVProDeckLink] Initialised (plugin v1.9.9 script v1.9.9)
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.DeckLinkManager:Awake()
UnityEngine.GameObject:AddComponent()
RenderHeads.Media.AVProDeckLink.Singleton`1:get_Instance()
RenderHeads.Media.AVProDeckLink.DeckLink:GetNumDevices()
<showDevicesINUM>d__52:MoveNext()
 
 

[AVProDeckLink] Using DeckLink API version 14.3.0.0
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.DeckLinkManager:Awake()
UnityEngine.GameObject:AddComponent()
RenderHeads.Media.AVProDeckLink.Singleton`1:get_Instance()
RenderHeads.Media.AVProDeckLink.DeckLink:GetNumDevices()
<showDevicesINUM>d__52:MoveNext()
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr)
CoroutineManager:startCoroutine(IEnumerator, Boolean)
 

constructing (singleton) RenderHeads.Media.AVProDeckLink.DeckLinkManager
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.Singleton`1:get_Instance()
RenderHeads.Media.AVProDeckLink.DeckLink:GetNumDevices()
<showDevicesINUM>d__52:MoveNext()
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr)
CoroutineManager:startCoroutine(IEnumerator, Boolean)
  
 
  

[AVProVideo] Initialising AVPro Video v2.9.3 (native plugin v2.8.5f1-core) on NVIDIA RTX A4000/Direct3D 11.0 [level 11.1] (MT False) on WindowsPlayer
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProVideo.MediaPlayer:Initialise()
RenderHeads.Media.AVProVideo.MediaPlayer:OpenMedia(MediaPathType, String, Boolean)
RenderHeads.Media.AVProVideo.MediaPlayer:OpenMedia(MediaPath, Boolean)
<playVideo>d__63:MoveNext()
 
   
constructing (singleton) RenderHeads.Media.AVProDeckLink.DeckLinkSettings
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.Singleton`1:get_Instance()
RenderHeads.Media.AVProDeckLink.DeckLinkManager:LateUpdate()
 

[10:07:12.061][CameraRecordMenuButtonController]showDevicesINUM DeckLink TOTALDEVICES  numDevices=0

WORKING

 

[AVProDeckLink] Initialised (plugin v1.9.7 script v1.9.7)
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.DeckLinkManager:Awake()
UnityEngine.GameObject:AddComponent()
RenderHeads.Media.AVProDeckLink.Singleton`1:get_Instance()
RenderHeads.Media.AVProDeckLink.DeckLink:GetNumDevices()
<showDevicesINUM>d__52:MoveNext()
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr)
 

[AVProDeckLink] Using DeckLink API version 12.5.1.0
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.DeckLinkManager:Awake()
UnityEngine.GameObject:AddComponent()
RenderHeads.Media.AVProDeckLink.Singleton`1:get_Instance()
RenderHeads.Media.AVProDeckLink.DeckLink:GetNumDevices()
<showDevicesINUM>d__52:MoveNext()
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr)
 
 

constructing (singleton) RenderHeads.Media.AVProDeckLink.DeckLinkManager
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.Singleton`1:get_Instance()
RenderHeads.Media.AVProDeckLink.DeckLink:GetNumDevices()
<showDevicesINUM>d__52:MoveNext()
UnityEngine.SetupCoroutine:InvokeMoveNext(IEnumerator, IntPtr)
CoroutineManager:startCoroutine(IEnumerator, Boolean)
RecordMenuButtonController:onChangeDropDown()
UnityEngine.Events.UnityEvent`1:Invoke(T0)
TMPro.TMP_Dropdown:SetValue(Int32, Boolean)
 

[10:14:05.641][CameraRecordMenuButtonController]showDevicesINUM TOTALDEVICES  numDevices=4

RichRH, 2025-01-24:
Ok, thanks for confirming it was 1.9.7 that worked - I seem to remember since that version we had to upgrade some scripts in our projects for more recent Unity versions.

As per my previous message, if it all possible could you please test our '04_InputExplorerDemo' in 1.9.9 to see if that works for you as it does for me? That could hopefully guide as to whether it's more a native code or script level issue.

Cheers,

richard12121212, 2025-01-29:
HI, sorry for the delay. The hardware was at the office.
Hereby some extra information:
when we do
DeckLinkPlugin.GetNumDevices it gives 4 devices. 

DeckLink.GetNumDevices();  it gives zero devices

We build your demo scene (01_InputDemo) which results in:

[PhysX] Initialized MultithreadedTaskDispatcher with 48 workers.
Initialize engine version: 2022.3.54f1 (129125d4e700)
[Subsystems] Discovering subsystems at path C:/Users/immer/Downloads/current2020/current2020/CoVince_Data/UnitySubsystems
GfxDevice: creating device client; threaded=1; jobified=1
Direct3D:
    Version:  Direct3D 11.0 [level 11.1]
    Renderer: NVIDIA RTX A4000 (ID=0x24b0)
    Vendor:   NVIDIA
    VRAM:     16169 MB
    Driver:   32.0.15.6603
The referenced script (Unknown) on this Behaviour is missing!
The referenced script on this Behaviour (Game Object '<null>') is missing!
<RI> Initializing input.
New input system (experimental) initialized
Using Windows.Gaming.Input
XInput1_3.dll not found. Trying XInput9_1_0.dll instead...
<RI> Input initialized.
<RI> Initialized touch support.
[3D WebView] Using Vuplex 3D WebView for Windows v4.10.1 
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
Vuplex.WebView.StandaloneWebView:_initializePlugin()

[PhysX] Initialized MultithreadedTaskDispatcher with 48 workers.
UnloadTime: 0.607800 ms
[3D WebView] System info (used by Vuplex support):
Unity version: 2022.3.54f1
Development build: False
OS version: Windows 11  (10.0.26100) 64bit
Device model: Amd Wrx80 (PCSpecialist)
Graphics API: Direct3D11
Rendering threading mode: LegacyJobified
Render pipeline: default
SRP Batcher: False
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
Vuplex.WebView.Internal.BaseWebView:_logSystemInfoIfNeeded()

[AVProDeckLink] DeckLinkPlugin.GetNumDevices()=4
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.DeckLinkManager:EnumDevices()
RenderHeads.Media.AVProDeckLink.DeckLinkManager:Init()
RenderHeads.Media.AVProDeckLink.DeckLinkManager:Awake()

[AVProDeckLink] Initialised (plugin v1.9.9 script v1.9.9)
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.DeckLinkManager:Awake()

[AVProDeckLink] Using DeckLink API version 14.3.0.0
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.DeckLinkManager:Awake()

[AVProDeckLink] DeckLinkPlugin.GetNumDevices()=4
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:Log(Object)
RenderHeads.Media.AVProDeckLink.DeckLinkManager:EnumDevices()
RenderHeads.Media.AVProDeckLink.DeckLinkManager:Init()
RenderHeads.Media.AVProDeckLink.DeckLinkSettings:Start()

[AVProDeckLink] Desired device not found
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogWarning(Object)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin(Boolean)
RenderHeads.Media.AVProDeckLink.DeckLink:Start()

[AVProDeckLink] Desired device not found
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogWarning(Object)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin(Boolean)
RenderHeads.Media.AVProDeckLink.DeckLink:RenderExplorer()
RenderHeads.Media.AVProDeckLink.DeviceExplorerManager:OnGUI()

[AVProDeckLink] Desired device not found
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogWarning(Object)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin(Boolean)
RenderHeads.Media.AVProDeckLink.DeckLink:RenderExplorer()
RenderHeads.Media.AVProDeckLink.DeviceExplorerManager:OnGUI()

[AVProDeckLink] Desired device not found
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogWarning(Object)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin(Boolean)
RenderHeads.Media.AVProDeckLink.DeckLink:RenderExplorer()
RenderHeads.Media.AVProDeckLink.DeviceExplorerManager:OnGUI()

[AVProDeckLink] Desired device not found
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogWarning(Object)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin(Boolean)
RenderHeads.Media.AVProDeckLink.DeckLink:RenderExplorer()
RenderHeads.Media.AVProDeckLink.DeviceExplorerManager:OnGUI()

[AVProDeckLink] Desired device not found
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogWarning(Object)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin(Boolean)
RenderHeads.Media.AVProDeckLink.DeckLink:RenderExplorer()
RenderHeads.Media.AVProDeckLink.DeviceExplorerManager:OnGUI()

[AVProDeckLink] Desired device not found
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogWarning(Object)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin(Boolean)
RenderHeads.Media.AVProDeckLink.DeckLink:RenderExplorer()
RenderHeads.Media.AVProDeckLink.DeviceExplorerManager:OnGUI()

[AVProDeckLink] Desired device not found
UnityEngine.Logger:Log(LogType, Object)
UnityEngine.Debug:LogWarning(Object)
RenderHeads.Media.AVProDeckLink.DeckLink:Begin(Boolean)
RenderHeads.Media.AVProDeckLink.DeckLink:RenderExplorer()
RenderHeads.Media.AVProDeckLink.DeviceExplorerManager:OnGUI()

RichRH, 2025-01-29:
Hi @richard12121212,

No worries and thanks for the extra info.

I'm struggling to reproduce this - Unity 2022.3.56f1, both latest code through debugger and the 1.9.9 Trial DLL I believe you're using, on the same DeckLink Quad HDMI Recorder and it just works (editor and build).

I'll keep digging but just to confirm - in the above test have you created a blank project, imported our package and opened our input test? (I see a bit of unfamiliar logging), and when you say 'We build your demo scene' are you actually creating builds or just testing in editor?

Cheers,

richard12121212, 2025-01-29:
Dear Richard,

We have a paid version which we import from the unity store. (so not trial)

It was not an entirely blank project, we used a base project where we put your scene to build. We can try in blank one, but my feeling is that this wont really change a lot.

it is indeed actual build, so not in editor.

RichRH, 2025-01-29:
Apologies, in your initial ticket it states Trial version.

If you could please test a fresh project just to discount any oddities (and match what I'm testing) that'd be great. Also, are the results in editor any different?

Cheers,

richard12121212, 2025-01-29:
We will build in new project to test. 
We do not have the card currently in PC with editor.

is there option to build some extra logging, into the dll?

richard12121212, 2025-01-29:
ps sorry , probably mist the Trial field. my mistake

RichRH, 2025-01-29:
Great, thanks and understood on no editor on the test PC.

There is actually a flag on the 'DeckLinkManager' component -> 'Log Device Enumeration' that _might_ provide a clue.

Cheers,

richard12121212, 2025-01-30:
The Log Device enumaration is in the unity C# file. this is only logging, after the getdevices.  
but i meant logging in the dll, so in this

		[DllImport("AVProDeckLink")]
		public static extern int GetNumDevices();

 PS. small thing, probably not related. but when building with plugin imported we get (we can solve this ofcourse but might be good to have it in package)
An asset is marked with HideFlags.DontSave but is included in the build:
Asset: 'Assets/AVProDeckLink/Resources/AVProDeckLinkIcon.png'
Asset name: AVProDeckLinkIcon
(You are probably referencing internal Unity data in your build.)
UnityEditor.BuildPlayerWindow:BuildPlayerAndRun ()

RichRH, 2025-01-30:
Thanks for testing that @richard12121212 - are you able to send that exact test project and build to me? (via unitysupport@renderheads.com)

(And noted on the build issue you mentioned above - will look into that separately)

Cheers,

RichRH, 2025-01-30:
Thanks for sending that through @richard12121212 - I've only been able to run on our 8K Pro machine here but works fine:

[image]

I'll be able to test on our DeckLink Quad HDMI Recorder machine this evening.

One other thought - have you tried Blackmagic's Media Express on that same machine to confirm input is as expected?

Cheers,

richard12121212, 2025-01-30:
i used the media express to inout is working, and as i mentioned above. wehn we go back to older version it works also with your pacakge.

RichRH, 2025-01-30:
Ok, checked your build on our DeckLink Quad HDMI Recorder machine, and rather frustratingly it of course also works fine!

[image]

[image]

Same build, same card, different results - just trying to fathom what else could be at play. When installing Desktop Video 14.3 it updated the firmware on our card - presume you had the same? I'm trying to locate any info. on fimware versions etc.. but the Desktop Video Setup software barely provides anything:

[image]

[image]

[image]

Might have to resort to further logging at native level but slighty tricky as not sure where it's falling over..

RichRH, 2025-02-25:
Hi @richard12121212,

Just wanted to check you saw my above question - was the card firmware successfully updated when 14.3 was installed? (It makes sense if that didn't happen/failed then the 14.3 APIs may not recognise the card whereas older versions might)

I'm back looking into this and potential logging options, but it seems logging things like firmware version don't seem supported so obvious routes for comparing working (mine) and non-working versions is trickier than hoped.

Cheers,